arduino/arduino-cli #1: Add a way to identify the current sketch and its location

tigoe, 2018-08-23:
board attach works OK, but users who are moving between a few tasks may forget which sketch they're on. Making arduino-cli sketch return the name and path of the current sketch would be handy.

cmaglie, 2018-09-20:
@tigoe may you expand a bit more on that?
I can see that the shell prompt already gives you the working directory that reveal also the sketch name, for example in a typical session I have:

```
cmaglie@crocchetta:~$ cd Arduino
cmaglie@crocchetta:~/Arduino$ ls
blink  libraries  xxxx  yyyyy
cmaglie@crocchetta:~/Arduino$ cd blink/
cmaglie@crocchetta:~/Arduino/blink$ 
cmaglie@crocchetta:~/Arduino/blink$ 
```

at this point is clear that I'm working on the `blink` sketch because the shell prompt just tell me the directory I'm currently in, so having another command like:

```
cmaglie@crocchetta:~/Arduino/blink$ arduino-cli sketch
Sketch: blink
Directory: /home/cmaglie/Arduino/blink
```

looks like a bit redundant.

tigoe, 2018-09-20:
You're assuming that I've cd'ed to the project directory before I start working. That's not always the case. Sometimes I might be working on a group of sketches all at once, kept in different directories.

cmaglie, 2018-09-21:
If you're working on multiple sketches you have to tell the CLI which one you're referring to.
The only case of "implicit" sketch (when you omit it) is the one in the current working directory.

Or maybe I'm totally misunderstanding your use case, if you can provide a transcript of a terminal session to show me it may be quicker a way :-)

tigoe, 2018-09-21:
I can't, because the transcript is not on my own machine :(. But I was working on a machine yesterday where the primary interface is a screen reader, not visual reading, and so therefore the owner of the machine customized the bash prompt so that, instead of giving the directory and therefore reading the whole long  prompt every time, it just gives $. Try turning on the VoiceOver screen reader on MacOS  and operating your computer that way, and you'll get a sense of it.  command-F5 will turn it on. 

Then, the person preferred to work out of the home directory, therefore requiring /Documents/Arduino/sketchName every time. tab helps, but it's still a pain in the butt. Likewise, arduino:avr:uno every time you want to upload is difficult. If there is a profile for the sketch, so that there's a default sketch directory for it, and a default board type, and port, then you can just type

$arduino-cli upload 

for repeated uploads instead of
cmaglie@crocchetta:~/Arduino/blink$  arduino-cli upload -p /dev/cu.usbmodem14111 -b arduino:samd:mkrzero ~/Documents/Arduino/cliTest/

Through the screen reader, my line reads in a second or two. Yours takes about 15 seconds, slowing development time considerably. 

Of course, I might forget what my default is, so being able to ask for it, maybe by: 

$arduino-cli board id

or 

$arduino-cli sketch location
 
would help a lot.

This is difficult to convey without a screen reader, but I am happy to get on hangouts with you and give you the full experience, Cristian.

cmaglie, 2018-09-21:
> customized the bash prompt so that, instead of giving the directory and therefore reading the whole long prompt every time, it just gives $

ok this is the detail that I was missing. So my transcript above, in your case, will look like:

```
$ cd Arduino
$ ls
blink  libraries  xxxx  yyyyy
$ cd blink/
$ 
$ 
[...and with the proposed sketch command...]
$ arduino-cli sketch
Sketch: blink
Directory: /home/cmaglie/Arduino/blink
```

and now your request makes perfect sense.
BTW the nice think with the shell is that is highly customizable, so you can get the same information **right now**:

```
$ pwd
/home/cmaglie/Arduino/blink
```

(pwd means "print working directory") or if you are interested in just the name of the sketch you can do something like:

```
$ basename /home/cmaglie/Arduino/blink
blink
$ basename $(pwd)
blink
$ alias sk="basename $(pwd)"
$ sk
blink
```

basename returns just the last part of the path, so you can program a shell alias called `sk` as in the example above, this will make your experience even more quicker than typing `arduino-cli sketch`.

tigoe, 2018-09-21:
I know how pwd works.... 

The issue here isn't knowing where I am, it's knowing where my sketch is, when I am not in the same directory. And doing so in a not-overly-verbose way. The two are not necessarily the same, You're assuming only one way of working with the cli, but bash supports coming at it from many directions.

cmaglie, 2018-09-21:
Ok, then I'm happy to do an hangout to see your workflow, drop me a line via email to coordinate.

tigoe, 2018-09-21:
Will do.

PR #1466 (2021-09-24): Fix library verification when installing from git repo or zip file
**Please check if the PR fulfills these requirements**

- [x] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-cli/pulls)
      before creating one)
- [x] The PR follows
      [our contributing guidelines](https://arduino.github.io/arduino-cli/latest/CONTRIBUTING/#pull-requests)
- [x] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)
- [ ] `UPGRADING.md` has been updated with a migration guide (for  [... cut]