arduino/arduino-cli #3: Simplify and add shortcuts for core workflow

tigoe, 2018-08-23:
The core workflow needs some simplification of its commands with some config additions. I apologize if these are already in the pipeline.

The core workflow for most sketches is something like this:

1. create sketch
2. edit sketch
3. choose board
4. choose port
5. verify and upload sketch to board

1, 2, and 4 are generally once-per-sketch tasks, while 2 and 5 are generally done repeatedly. 

Right now, it looks like this:
```
$ arduino-cli sketch new cliSketch
$ open ~/Documents/Arduino/cliSketch/cliSketch.ino  # open is a placeholder for your favorite editor
$ arduino-cli board list
$ arduino-cli  board attach /dev/cu.usbmodem14121 cliSketch/cliSketch.ino
$arduino-cli compile --fqbn arduino:avr:uno ~/Documents/Arduino/cliSketch/
$ arduino-cli upload --fqbn arduino:avr:uno -p /dev/cu.usbmodem14121 ~/Documents/Arduino/cliSketch2/
```

This can get shorter if you use the menus from the GUI as a guide (loosely), and use some shortcut names. For example:

$ arduino-cli sketch new cliSketch  should set the current working directory for the cli, so that cliSketch is auto-expanded in any command that uses it. shortcut: s -n

$ arduino-cli sketch with no parameters could list the current sketch and CWD

$ arduino-cli port set /dev/cu.usbmodem14121 could set the working port for the cli, or for this sketch Follows the tools --> port menu behavior shortcut p -s

$ arduino-cli port with no parameters could list the current port

$ arduino-cli board set arduino:avr:uno could set the working board for the cli or for this sketch. Follows the tools --> board menu behavior. Shortcut: b -s. 

$ arduino-cli board with no parameters could list the current board

$ arduino-cli verify, with the current working directory set, could mirror the action of the current sketch --> verify/Compile, and be an alias for compile

$ arduino-cli upload, with the current working directory, board, and port set could mirror the action of the current sketch --> verify and upload 

If either verify or upload are called without board or port set, the cli should generate an appropriate error of course.

ThePerkinrex, 2018-08-30:
But if you have already set the port for a sketch, it could assume it if no port specified (and give an error if it doesn't find it).  And if it finds the port and knows the board on the port it could assume that as well if not specified

tigoe, 2019-09-15:
Thanks, checking now. A couple things:  Check globally for a misspelling. You've got "Alredy" and it should be "Already".

I plugged in a MKR1000 and tried board list, and I got:

````
Error detecting boards: error getting board info from Arduino Cloud: error querying Arduino Cloud Api: Get https://builder.arduino.cc/v3/boards/byVidPid//: net/http: TLS handshake timeout
````
I appear to have solved it by running listall and then running list. From there, I created a new sketch and cd'ed into the sketch to edit it. Then I did this:

````
tigoe$ arduino-cli board attach /dev/cu.usbmodem141301 
panic: runtime error: index out of range

goroutine 1 [running]:
github.com/arduino/arduino-cli/cli/board.runAttachCommand(0xc0001bf180, 0xc0000a1590, 0x1, 0x1)
	/__w/arduino-cli/arduino-cli/cli/board/attach.go:57 +0x205
github.com/spf13/cobra.(*Command).execute(0xc0001bf180, 0xc0000a1550, 0x1, 0x1, 0xc0001bf180, 0xc0000a1550)
	/github/home/go/pkg/mod/github.com/spf13/cobra@v0.0.5/command.go:830 +0x2ae
github.com/spf13/cobra.(*Command).ExecuteC(0x4ea3300, 0x4696f51, 0xc0001f9f88, 0xc000096058)
	/github/home/go/pkg/mod/github.com/spf13/cobra@v0.0.5/command.go:914 +0x2fc
github.com/spf13/cobra.(*Command).Execute(...)
	/github/home/go/pkg/mod/github.com/spf13/cobra@v0.0.5/command.go:864
main.main()
	/__w/arduino-cli/arduino-cli/main.go:28 +0x2e
tigoeMacBook130:cliSketch tigoe$ 
````
What am I missing?

I think there is need for a more verbose guide than just the man file.

tigoe, 2019-09-15:
OK, after much trial and error, here's the closest I could get to what you describe. I couldn't attach a configuration (port and board) to the sketch because I couldn't figure out how:

````
 tigoe$ arduino-cli board list
Port                   Type              Board Name              FQBN                
/dev/cu.usbmodem141301 Serial Port (USB) Arduino/Genuino MKR1000 arduino:samd:mkr1000

tigoeMacBook130:Arduino tigoe$ arduino-cli compile --fqbn arduino:samd:mkr1000 cliSketch/
Sketch uses 10916 bytes (4%) of program storage space. Maximum is 262144 bytes.
tigoeMacBook130:Arduino tigoe$ arduino-cli upload --fqbn arduino:samd:mkr1000 -p /dev/cu.usbmodem141301 cliSketch/
No new serial port detected.
Error during Upload: cannot execute upload tool: fork/exec {runtime.tools.bossac-1.7.0-arduino3.path}/bossac: no such file or directory
tigoeMacBook130:Arduino tigoe$ 
````

tigoe, 2019-09-15:
OK, I got a bit closer. We're still not there, but closer:

````
tigoe$ arduino-cli sketch new cliSketch
Sketch created in: /Users/tigoe/Documents/Arduino/cliSketch
tigoeMacBook130:Arduino tigoe$ arduino-cli board list
Port                   Type              Board Name              FQBN                
/dev/cu.usbmodem141301 Serial Port (USB) Arduino/Genuino MKR1000 arduino:samd:mkr1000

tigoeMacBook130:Arduino tigoe$ arduino-cli board attach arduino:samd:mkr1000 cliSketch/
Selected fqbn: arduino:samd:mkr1000
tigoeMacBook130:Arduino tigoe$ arduino-cli board attach /dev/cu.usbmodem141301 cliSketch/
Board found: Arduino/Genuino MKR1000...
Selected fqbn: arduino:samd:mkr1000
tigoeMacBook130:Arduino tigoe$ arduino-cli upload cliSketch/
Error during Upload: no upload port provided
tigoeMacBook130:Arduino tigoe$ arduino-cli upload -p /dev/cu.usbmodem141301 cliSketch/
Error during Upload: compiled sketch cliSketch/cliSketch.arduino.samd.mkr1000.bin not found
tigoeMacBook130:Arduino tigoe$ arduino-cli compile cliSketch/
Sketch uses 10916 bytes (4%) of program storage space. Maximum is 262144 bytes.
tigoeMacBook130:Arduino tigoe$ arduino-cli upload -p /dev/cu.usbmodem141301 cliSketch/
No new serial port detected.
Error during Upload: cannot execute upload tool: fork/exec {runtime.tools.bossac-1.7.0-arduino3.path}/bossac: no such file or directory

````

per1234, 2021-02-03:
Closing as resolved.

There is a somewhat related feature request for `arduino-cli board attach` at https://github.com/arduino/arduino-cli/issues/1073, but that wouldn't be necessary for the MKR 1000 used as an example here since it can be auto-detected.

closed 2021-02-03 by per1234