arduino/arduino-cli #5: How to add URL for other boards?

gdsports, 2018-08-25:
I see board_manager in .cli_config.yml but I cannot figure out the syntax to add a board URL for Adafruit or ESP8266.

The IDE can do this from the command line like this. I am converting a script using the IDE command line to the new arduino-cli.

`./arduino --pref "boardsmanager.additional.urls=https://adafruit.github.io/arduino-board-index/package_adafruit_index.json" --save-prefs
`

machadolab, 2018-08-25:
I have the same question. I tried editing ./.cli-config.yml manually to edit the ```board_manager``` setting, after saving that and running ```arduino-cli core update-index``` and then searching for a new core, no core from the new URL is found.

laleporo, 2018-08-27:
You can find the correct config names in yaml_serializer.go: it's `additional_urls`.
So to add the esp board, you have to change `board_manager: null` to

    board_manager:
      additional_urls:
        - http://arduino.esp8266.com/stable/package_esp8266com_index.json

gdsports, 2018-08-27:
Thanks, I tried it and it works fine.

closed 2018-08-28 by cmaglie

rogercm, 2018-11-01:
Hi all. Could someone give detailed tips for a Arduino newbie on how to add a nodemcu (esp12-e) using the new arduino-cli? Using IDE is easy, but I'm totally lost with the new "cli". Thanks in advance!