arduino/arduino-cli #6: Poorly behaved on environments which don't already have arduino-ide

shaunoneil, 2018-08-25:
Arduino-cli appears to assume the IDE exists, and doesn't always handle the lack cleanly.

Example (release binary arduino-cli-0.2.0-alpha.preview-osx):

    $ arduino-cli board listall
    Error: loading json index file /Users/soneil/Library/arduino15/package_index.json: open /Users/soneil/Library/arduino15/package_index.json: no such file or directory
    Failed to load https://downloads.arduino.cc/packages/package_index.json package index.
    Try updating all indexes with `arduino-cli core update-index`.

    $ arduino-cli core update-index
    Updating index: package_index.json downloaded
    Error: open /Users/soneil/Library/arduino15/package_index.json: no such file or directory
    Error saving downloaded index https://downloads.arduino.cc/packages/package_index.json

In this case, `~/Library/arduino15` is assumed to exist, and is not created if it doesn't.  Manually creating `~/Library/arduino15` allows update-index to complete cleanly.

flatsiedatsie, 2018-12-14:
I also just ran into this.

I followed the instructions here (version 0.2.2)

https://forum.mysensors.org/topic/9845/arduino-has-an-official-cli-app-now/2

But something went wrong on the update command.

```
pi@gateway:~/uploader $ ./arduino-cli-0.3.2-alpha.preview-linuxarm sketch new MyFirstSketch
Sketch created in: /home/pi/Arduino/MyFirstSketch
pi@gateway:~/uploader $ ./arduino-cli-0.3.2-alpha.preview-linuxarm core update-index       
 0 / ? [------------------------------------------------------------------------------------------=]   0.00%panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x4 pc=0x123d4]

goroutine 14 [running]:
sync/atomic.addUint64(0x113c4194, 0x1000, 0x0, 0x1000, 0x1000)
	/home/jenkins/go1.10.2-patched/src/sync/atomic/64bit_arm.go:31 +0x4c
github.com/arduino/arduino-cli/vendor/go.bug.st/downloader.(*Downloader).AsyncRun(0x113c4180)
	/home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/go.bug.st/downloader/downloader.go:82 +0x17c
created by github.com/arduino/arduino-cli/vendor/go.bug.st/downloader.(*Downloader).RunAndPoll
	/home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/go.bug.st/downloader/downloader.go:61 +0x70
pi@gateway:~/uploader $ ./arduino-cli-0.3.2-alpha.preview-linuxarm board list       
Error: loading json index file /home/pi/.arduino15/package_index.json: open /home/pi/.arduino15/package_index.json: no such file or directory
Failed to load https://downloads.arduino.cc/packages/package_index.json package index.
Try updating all indexes with `arduino-cli-0.3.2-alpha.preview-linuxarm core update-index`.
```

per1234, 2018-12-15:
@flatsiedatsie you're using an outdated version of arduino-cli. That [bug ](https://github.com/arduino/arduino-cli/issues/87)was already [fixed](https://github.com/arduino/arduino-cli/pull/91) after the 0.3.2-alpha.preview release. You need to install the latest version (0.3.3-alpha.preview).

flatsiedatsie, 2018-12-17:
Thanks. I updated the instructions on the MySensors forum to use the 0.3.3 version.

commit 2fa46edd (2018-09-14): Create data dir if not found on 'core update-index'

Fix #6