arduino/arduino-cli #8: arduino-cli OSX doesn't create arduino15 folder

fabiopozzi, 2018-08-25:
I have downloaded the arduino-cli binary, installed in `/usr/local/bin/`.
When I run `arduino-cli core update-index` for the first time I get
```
Updating index: package_index.json downloaded
Error: open /Users/fabio/Library/arduino15/package_index.json: no such file or directory
Error saving downloaded index https://downloads.arduino.cc/packages/package_index.json
```
So I manually create the folder:
```
mkdir /Users/fabio/Library/arduino15
```
and the issue is solved.
I don't know if the documentation should indicate to the user to create the directory or the tool should create it.
I am willing to help patch the sources if the tool should create the directory.

michelepagot, 2018-08-25:
Duplicare of issue 6#

fabiopozzi, 2018-08-25:
I wonder if `getDefaultArduinoDataDir()` should create the directory if it doesn't exist or it should be created in `NewConfiguration()`.

sandeepmistry, 2018-08-29:
Closing as duplicate of #6.

closed 2018-08-29 by sandeepmistry