arduino/arduino-cli #12: Arduino nRF52 board from vendor velleman unkown board ?

cybernout, 2018-08-26:
does not know the arduino uno board from vendor velleman,

output From Arduino Ide :

Board Info:

BN: Arduino/Genuino Uno
VID: 2341
PID: 0043
SN: 856343138393515001E0

ouput from arduino-cli ( 64 bit linux , ubuntu 18.04 LTS ):

./arduino-cli board list
FQBN    Port        	ID       	Board Name
    	/dev/ttyACM0	2341:0043	unknown   

so compile fails as well output :

./arduino-cli compile --fqbn arduino:nrf52 /home/yellabs/Arduino/MySketch
Fully Qualified Board Name has incorrect format.

core files are present :

./arduino-cli core list
ID                	Installed	Latest	Name                
arduino:nrf52     	1.0.2    	1.0.2 	Arduino nRF52 Boards
sandeepmistry:nRF5	0.5.1    	0.5.1 	                 

please take a look and see if you can give advise or fix

cybernout, 2018-08-26:
i leave it here if some one else has this issue, but you could remove it, its solved by updating the arduino IDE , output now ,

./arduino-cli core list
ID                	Installed	Latest	Name                
arduino:avr       	1.6.21   	1.6.21	Arduino AVR Boards  
arduino:nrf52     	1.0.2    	1.0.2 	Arduino nRF52 Boards
sandeepmistry:nRF5	0.5.1    	0.5.1 	
                    
i can then compile without any problems to , 

./arduino-cli compile --fqbn arduino:avr:uno /home/yellabs/Arduino/MySketch
Downloading missing tool: builtin:ctags@5.8-arduino11
builtin:ctags@5.8-arduino11 downloaded                                                                                                                 
Installing builtin:ctags@5.8-arduino11
Sketch uses 928 bytes (2%) of program storage space. Maximum is 32256 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 2039 bytes for local variables. Maximum is 2048 bytes.

closed 2018-08-26 by cybernout

JoannaBro, 2020-11-04:
I had the same problem. The `arduino:nrf52` platform supports only one board - Arduino Primo and to use it, you should call the following command: 
`$ ./arduino-cli compile --fqbn arduino:nrf52:primo /home/yellabs/Arduino/MySketch`.