arduino/arduino-cli #14: Support portable folder workflow for local installation

PradeepMW, 2018-08-27:
Similar to portable folder workflow in the main IDE, provide a similar option in the command line interface.

cmaglie, 2018-08-30:
There are currently two way to achieve this.

1) Create your `.cli-config.yml` file with the following content:

```
sketchbook_path: /home/cmaglie/Workspace/sketchbook-cores-beta
arduino_data: /home/cmaglie/.arduino15
```

and set `sketchbook_path` and `arduino_data` to your preferred location.
You can check if the CLI is reading the config with the command:

```
$ arduino-cli config dump
proxy_type: auto
sketchbook_path: /home/cmaglie/Workspace/sketchbook-cores-beta
arduino_data: /home/cmaglie/.arduino15
```

(you can also add `--debug` if it's not taking your config file to see what's happening)

2) you can use the env variables `ARDUINO_DATA_DIR` and `ARDUINO_SKETCHBOOK_DIR`:

```
$ ARDUINO_DATA_DIR=~/arduino/data ARDUINO_SKETCHBOOK_DIR=~/arduino/sketch arduino-cli config dump
sketchbook_path: /home/cmaglie/arduino/sketch
arduino_data: /home/cmaglie/arduino/data
```

masci, 2019-08-05:
I'm closing this, feel free to join the conversation here: https://github.com/arduino/arduino-cli/issues/108

closed 2019-08-05 by masci