arduino/arduino-cli #15: 'core list' doesn't work if a core is installed in sketchbook/hardware folder

manchoz, 2018-08-27:
for example:

```
$ arduino-cli core --debug list
INFO[0000] Initiating configuration
INFO[0000] Unserializing configurations from /usr/local/bin/.cli-config.yml
WARN[0000] Error reading config, using default configuration  error="open /usr/local/bin/.cli-config.yml: no such file or directory"
WARN[0000] Did not manage to get config file, using default configuration  error="open /usr/local/bin/.cli-config.yml: no such file or directory"
INFO[0000] Checking if CLI is Bundled into the IDE
INFO[0000] Candidate IDE Directory: /usr/local/bin
INFO[0000] CLI is not bundled into the IDE
INFO[0000] Configuration set
INFO[0000] Starting root command preparation (`arduino`)
INFO[0000] Formatter set
INFO[0000] Executing `arduino core list`
INFO[0000] Loading the default Package index
INFO[0000] Loading hardware from: /Users/mancho/Library/arduino15/packages
INFO[0000] Loading package arduino from: /Users/mancho/Library/arduino15/packages/arduino/hardware
INFO[0000] Loaded platform                               platform="arduino:avr@1.6.21"
INFO[0000] Loaded platform                               platform="arduino:samd@1.6.19"
INFO[0000] Loaded platform                               platform="arduino:samd_beta@1.6.22"
INFO[0000] Checking existence of 'tools' path: /Users/mancho/Library/arduino15/packages/arduino/tools
INFO[0000] Loading tools from dir: /Users/mancho/Library/arduino15/packages/arduino/tools
INFO[0000] Loaded tool                                   tool="arduino:CMSIS@4.5.0"
INFO[0000] Loaded tool                                   tool="arduino:CMSIS-Atmel@1.1.0"
INFO[0000] Loaded tool                                   tool="arduino:CMSIS-Atmel@1.2.0"
INFO[0000] Loaded tool                                   tool="arduino:arduinoOTA@1.1.1"
INFO[0000] Loaded tool                                   tool="arduino:arduinoOTA@1.2.0"
INFO[0000] Loaded tool                                   tool="arduino:arm-none-eabi-gcc@4.8.3-2014q1"
INFO[0000] Loaded tool                                   tool="arduino:arm-none-eabi-gcc@7-2017q4"
INFO[0000] Loaded tool                                   tool="arduino:avr-gcc@4.9.2-atmel3.5.4-arduino2"
INFO[0000] Loaded tool                                   tool="arduino:avrdude@6.3.0-arduino9"
INFO[0000] Loaded tool                                   tool="arduino:bossac@1.7.0"
INFO[0000] Loaded tool                                   tool="arduino:bossac@1.7.0-arduino3"
INFO[0000] Loaded tool                                   tool="arduino:openocd@0.10.0-arduino7"
INFO[0000] Loaded tool                                   tool="arduino:openocd@0.9.0-arduino6-static"
INFO[0000] Loading package builtin from: /Users/mancho/Library/arduino15/packages/builtin
INFO[0000] Checking existence of 'tools' path: /Users/mancho/Library/arduino15/packages/builtin/tools
INFO[0000] Loading tools from dir: /Users/mancho/Library/arduino15/packages/builtin/tools
INFO[0000] Loaded tool                                   tool="builtin:ctags@5.8-arduino11"
INFO[0000] Loading hardware from: /Users/mancho/Documents/Arduino/hardware
INFO[0000] Loading package arduino-new from: /Users/mancho/Documents/Arduino/hardware/arduino-new
INFO[0000] Loaded platform                               platform="arduino-new:samd@"
INFO[0000] Loading package esp8266com from: /Users/mancho/Documents/Arduino/hardware/esp8266com
INFO[0000] Loaded platform                               platform="esp8266com:esp8266@"
INFO[0000] Loading package espressif from: /Users/mancho/Documents/Arduino/hardware/espressif
INFO[0000] Loaded platform                               platform="espressif:esp32@"
%!s(PANIC=runtime error: invalid memory address or nil pointer dereference)
```

tttapa, 2018-08-28:
I have the same problem on Ubuntu 16.04.

```
$ ./arduino-cli-0.2.0-alpha.preview-linux64 --debug core list
INFO[0000] Initiating configuration                     
INFO[0000] Unserializing configurations from /home/pieter/Downloads/.cli-config.yml 
WARN[0000] Error reading config, using default configuration  error="open /home/pieter/Downloads/.cli-config.yml: no such file or directory"
WARN[0000] Did not manage to get config file, using default configuration  error="open /home/pieter/Downloads/.cli-config.yml: no such file or directory"
INFO[0000] Checking if CLI is Bundled into the IDE      
INFO[0000] Candidate IDE Directory: /home/pieter/Downloads 
INFO[0000] CLI is not bundled into the IDE              
INFO[0000] Configuration set                            
INFO[0000] Starting root command preparation (`arduino`) 
INFO[0000] Formatter set                                
INFO[0000] Executing `arduino core list`                
INFO[0000] Loading the default Package index            
INFO[0000] Loading hardware from: /home/pieter/.arduino15/packages 
INFO[0000] Loading package ATTinyCore from: /home/pieter/.arduino15/packages/ATTinyCore/hardware 
INFO[0000] Loaded platform                               platform="ATTinyCore:avr@1.1.5"
INFO[0000] Loading package SparkFun from: /home/pieter/.arduino15/packages/SparkFun/hardware 
INFO[0000] Loaded platform                               platform="SparkFun:avr@1.1.11"
INFO[0000] Loading package arduino from: /home/pieter/.arduino15/packages/arduino/hardware 
INFO[0000] Loaded platform                               platform="arduino:avr@1.6.20"
INFO[0000] Loaded platform                               platform="arduino:sam@1.6.11"
INFO[0000] Checking existence of 'tools' path: /home/pieter/.arduino15/packages/arduino/tools 
INFO[0000] Loading tools from dir: /home/pieter/.arduino15/packages/arduino/tools 
INFO[0000] Loaded tool                                   tool="arduino:arduinoOTA@1.1.1"
INFO[0000] Loaded tool                                   tool="arduino:arm-none-eabi-gcc@4.8.3-2014q1"
INFO[0000] Loaded tool                                   tool="arduino:avr-gcc@4.9.2-atmel3.5.4-arduino2"
INFO[0000] Loaded tool                                   tool="arduino:avrdude@6.3.0-arduino9"
INFO[0000] Loaded tool                                   tool="arduino:bossac@1.6.1-arduino"
INFO[0000] Loading package esp8266 from: /home/pieter/.arduino15/packages/esp8266/hardware 
INFO[0000] Loaded platform                               platform="esp8266:esp8266@2.4.1"
INFO[0000] Checking existence of 'tools' path: /home/pieter/.arduino15/packages/esp8266/tools 
INFO[0000] Loading tools from dir: /home/pieter/.arduino15/packages/esp8266/tools 
INFO[0000] Loaded tool                                   tool="esp8266:esptool@0.4.13"
INFO[0000] Loaded tool                                   tool="esp8266:mkspiffs@0.2.0"
INFO[0000] Loaded tool                                   tool="esp8266:xtensa-lx106-elf-gcc@1.20.0-26-gb404fb9-2"
INFO[0000] Loading hardware from: /home/pieter/Arduino/hardware 
INFO[0000] Loading package espressif from: /home/pieter/Arduino/hardware/espressif 
INFO[0000] Loaded platform                               platform="espressif:esp32@"
%!s(PANIC=runtime error: invalid memory address or nil pointer dereference)
```

cmaglie, 2018-08-28:
Fixed in 69d6e0ab2d379f003870690cf88191ada75a7ac2

closed 2018-08-28 by cmaglie

animeshitsme, 2018-08-29:
@manchoz can you provide me tool, tool="builtin:ctags@5.8-arduino11"
[image]

cmaglie, 2018-08-29:
@animeshitsme this is a bug in the 0.2.0-alpha.preview, please wait for the next alpha release or build it from source.

commit 69d6e0ab (2018-08-28): Fixed 'core list' command for platforms installed in the sketchbook

Fix #15