arduino/arduino-cli #17: Trying to run Arduino CLI on Raspberry pi

FernandoGarcia, 2018-08-28:
Hi!

I have successfully installed Go compiler and Arduino CLI using this steps:

```
wget https://dl.google.com/go/go1.11.linux-armv6l.tar.gz

sudo tar -C /usr/local -xvf go1.11.linux-armv6l.tar.gz

cat >> ~/.bashrc << 'EOF'

export GOPATH=$HOME/go

export PATH=/usr/local/go/bin:$PATH:$GOPATH/bin

EOF

source ~/.bashrc

go version
```

> result: go1.11 linux/arm

Then:

`go get -u github.com/arduino/arduino-cli`

Now following the https://github.com/arduino/arduino-cli#getting-started starts the problems.

`arduino-cli sketch new MyFirstSketch`

```
Returns: Error: mkdir /home/pi/Arduino/MyFirstSketch: permission denied
Could not create sketch directory.
```

So I tried using sudo the result is: 

> sudo: arduino-cli: command not found

To try solved this issue I have created the folder manually using

`sudo mkdir -p /home/pi/Arduino/MyFirstSketch`

On next step I could not download **package_index.json** due permissions error again.

So I did:

```
sudo mkdir -p /home/pi/.arduino15/

cd /home/pi/.arduino15/

wget https://downloads.arduino.cc/packages/package_index.json
```

After I have successfully installed AVR core using:

`arduino-cli core install arduino:avr`

The result for  `arduino-cli core list` is: 

> ID              Installed       Latest  Name
> arduino:avr     1.6.21          1.6.21  Arduino AVR Boards

Now using  `arduino-cli board list` the result is always: 

> FQBN    Port            ID              Board Name
>         /dev/ttyUSB0    0403:6001       unknown

In this point I don't know what to do to go on since the CLI doesn't find a FQBN to my Arduino mega connected to USB port.

Can someone help please?

Best regards.

FernandoGarcia, 2018-08-28:
I don't know if it helps but the output for `lsusb` is: 

> Bus 001 Device 004: ID 0403:6001 Future Technology Devices International, Ltd FT232 USB-Serial (UART) IC

So I'm not using Arduino drivers.

FernandoGarcia, 2018-08-28:
Hi!

Thanks for your reply.

But what's the right value for Arduino Mega 2560 in this line for the argument --fqbn:

`arduino-cli compile --fqbn arduino:samd:mkr1000 Arduino/MyFirstSketch`

arduino:avr:mega2560 by example?

Best regards.

FernandoGarcia, 2018-08-28:
For `arduino-cli compile --fqbn arduino:avr:mega Arduino/MyFirstSketch`

The result is:

> avr-g++: error: missing device or architecture after '-mmcu='
> Error: exit status 1
> Compilation failed.
>

sandeepmistry, 2018-08-28:
@FernandoGarcia please try:

```
arduino-cli compile --fqbn arduino:avr:mega:cpu=atmega2560 Arduino/MyFirstSketch
```

Since the Mega has additional options (see Java IDE menu options for it), the FQBN needs the cpu type specified.

FernandoGarcia, 2018-08-29:
Thanks for your reply.

The settings are right now but I have permissions problem again.

`arduino-cli compile --fqbn arduino:avr:mega:cpu=atmega2560 Arduino/MyFirstSketch`

> Build options changed, rebuilding all
> Sketch uses 1462 bytes (0%) of program storage space. Maximum is 253952 bytes.
> Global variables use 9 bytes (0%) of dynamic memory, leaving 8183 bytes for local variables. Maximum is 8192 bytes.
> Error: open Arduino/MyFirstSketch/MyFirstSketch.arduino.avr.mega.cpu=atmega2560.hex: permission denied
> Error copying output file.

`sudo arduino-cli compile --fqbn arduino:avr:mega:cpu=atmega2560 Arduino/MyFirstSketch`

> sudo: arduino-cli: command not found

closed 2018-08-29 by FernandoGarcia