arduino/arduino-cli #20: arduino-cli core install arduino:avr failing on Windows 10

mancusoa74, 2018-08-28:
Operating System:

Edition: Windows 10 Enterprise
Version: 1709
OS Build: 16299.611
System Type: 64-bit operating system, x64-based processor

I have issued the following command:

arduino-cli-0.2.0-alpha.preview-windows.exe core install arduino:avr

and got this message:

The tool arduino:avr-gcc@4.9.2-atmel3.5.4-arduino2 is not available for the current OS

this prevents me to use the arduino-cli on my Windows system

Same command issued on Linux system (on same machine) works properly

animeshitsme, 2018-08-29:
I too have the same problem. Whatever I try to download in arduino cli it says that its not supported on your os.
Edition:- Windows 10 Home.

closed 2018-08-29 by cmaglie

luccasmaso, 2019-04-14:
Hi @cmaglie 

I'm still getting this. My environment of execution is `arm64` with `linux alpine`

```
Updating index: package_index.json downloaded 0.00%
FQBN    Port    ID      Board Name

Searching for platforms matching 'arduino'

ID                      Version Name                                             
Intel:arc32             2.0.4   Intel Curie Boards                               
arduino:avr             1.6.23  Arduino AVR Boards                               
arduino:megaavr         1.6.26  Arduino megaAVR Boards                           
arduino:nrf52           1.0.2   Arduino nRF52 Boards                             
arduino:sam             1.6.12  Arduino SAM Boards (32-bits ARM Cortex-M3)       
arduino:samd            1.6.21  Arduino SAMD Boards (32-bits ARM Cortex-M0+)     
arduino:samd_beta       1.6.25  Arduino SAMD Beta Boards (32-bits ARM Cortex-M0+)
arduino:stm32f4         1.0.1   Arduino STM32F4 Boards                           
littleBits:avr          1.0.0   littleBits Arduino AVR Modules                   

The tool arduino:avr-gcc@5.4.0-atmel3.6.1-arduino2 is not available for the current OS
FQBN    Port    ID      Board Name

Downloading and installing missing tool: builtin:ctags@5.8-arduino11
Error: tool not available for your OS
Error downloading builtin:ctags@5.8-arduino11
Error: platform arduino:avr is not installed
Invalid FQBN.
```

muellerj, 2019-11-20:
I am also having trouble. I'm running `Alpine` inside `WSL` on Windows10. The executables that are installed seem to be not the right architecture:

``` sh
$ ~/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++
Failed to execute process '/home/muellerj/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++'. Reason:
The file '/home/muellerj/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++' does not exist or could not be executed.
```

but the file is there:

``` sh
$ ls -a ~/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++
-rwxr-xr-x 2 muellerj muellerj 931K Nov 19 15:14 /home/muellerj/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-g++
```

cmaglie, 2019-11-20:
@luccasmaso @muellerj 
I'm not sure your problems are related to the above.

What is Alpine?

Please open a new issue, try with the latest `arduino-cli` and explain how to reproduce your problem.

muellerj, 2019-11-20:
It's a Linux distro: https://www.alpinelinux.org

mastrolinux, 2019-11-21:
The issue is all about WSL, not alpine itself.
The problem is WSL appears to be Linux as an OS but the underly OS is Windows. So best not to use WSL at all. Also WSL does not have access to serial ports of the system, so I strongly discourage using it. 
The Windows build of the CLI is not affected by this issue.

The same works on Ubuntu WSL which is officially supported by Windows while Alpine mess the permissions a bit.

```
luca@desktop:~$ arduino-cli core install arduino:avr
Downloading packages...
arduino:avr-gcc@5.4.0-atmel3.6.1-arduino2 already downloaded
arduino:avrdude@6.3.0-arduino14 already downloaded
arduino:arduinoOTA@1.2.1 already downloaded
arduino:avr@1.6.23 already downloaded
Installing arduino:avr-gcc@5.4.0-atmel3.6.1-arduino2...
arduino:avr-gcc@5.4.0-atmel3.6.1-arduino2 installed
Installing arduino:avrdude@6.3.0-arduino14...
arduino:avrdude@6.3.0-arduino14 installed
Installing arduino:arduinoOTA@1.2.1...
arduino:arduinoOTA@1.2.1 installed
Installing arduino:avr@1.6.23...
arduino:avr@1.6.23 installed
```

I would avoid supporting non-official builds of WSL.

Can you please tell us how did you install Alpine and a step by step guide on how to reproduce the issue? So I can check a bit more.

cmaglie, 2019-11-22:
I think that this deserve it's own issue, the original poster issue has been fixed long ago so I'm locking this one.

Feel free to open a new issue with a better description of the problem.

commit 1cd913b0 (2018-08-29): Fixed tool flavor detection for 32-bits systems

Fixes #20
Fixes #25
commit ff09ad1a (2019-07-16): Fixed tool flavor detection for 32-bits systems

Fixes #20
Fixes #25