arduino/arduino-cli #25: Missing tool:- builtin:ctags@5.8-arduino11

animeshitsme, 2018-08-29:
Does anyone have this tool:- builtin:ctags@5.8-arduino11.
While using arduino cli on my Windows I was able to create sketch but after that when I tried to compile it gives me an error as shown below.
I am not getting that anywhere on the internet or wasn't able to find it.
So it will be very generous of you all if you could provide me that tool.
Thank you in advance.
[image]

closed 2018-08-29 by cmaglie

manuel-covas, 2018-11-15:
Did you manage to add it manually? I have the same issue but I'm trying to get the cli to run in termux on arm64 Android.
Managed to build from source after bypassing OS name checks.

DMaisterra, 2022-03-28:
> Did you manage to add it manually? I have the same issue but I'm trying to get the cli to run in termux on arm64 Android. Managed to build from source after bypassing OS name checks.

oh men.... you can do a tutorial in this facts, you made it. or in case to bring any help. looking to make the same. would be good starting from thereq

commit 1cd913b0 (2018-08-29): Fixed tool flavor detection for 32-bits systems

Fixes #20
Fixes #25
commit ff09ad1a (2019-07-16): Fixed tool flavor detection for 32-bits systems

Fixes #20
Fixes #25