arduino/arduino-cli #27: External Programmer Support

vifino, 2018-08-30:
Hi.

First of all, I highly appreciate arduino-cli. It's a very, very big step forward in integrating into my development workflow, especially when you don't always have a graphical session running.
Not having Java around is a nice thing as well.

I do miss support for external programmers like a USBasp and friends, however.
While the Arduino bootloader is quite handy, sometimes it's not an option.

While I might have missed the option, I only saw the option to specify a port, not a programmer.

a7hybnj2, 2019-02-19:
This would be amazing! I currently use the IDE to upload via programmer then save the avrdude command needed to do so. I then make an .sh file in my project to use for that particular board and point it to the output of arduino-cli compile.

matthijskooijman, 2020-03-22:
When this is implemented, it might be good to clean up how these programmers are supported. I was looking through [the upload code in the Java IDE](https://github.com/arduino/Arduino/blob/master/arduino-core/src/cc/arduino/packages/uploaders/SerialUploader.java) today, and noticed it is a bit of a mess. In particular, it seems the way preferences of various parts (board platform, core platform, tool platform, programmer, board, tool) are loaded is not quite consistent between normal upload, programmer upload and burn bootloader. It would probably be good to clean this up and make it consistent between these, and also with the build processs, to make things a bit more predictable (e.g. now, upload using programmer does *not* use values from the platform.txt, while compilation does...).

ubidefeo, 2020-03-23:
@matthijskooijman don't worry about what goes on in the Java IDE too much.
CLI is built from the ground up and using external programmers is in development, just can't give you an ETA but it won't be long :)
We want to be sure we do this the right way and think twice to implement once 🙏

matthijskooijman, 2020-03-23:
Well, overall it *is* a good idea to implement things in the same way the IDE did them, to keep compatibility. In this case, I think it would be good to somewhat break compatibility in favor of consistency, so just wanted to make that explicit.

Good to hear that there is some work ongoing. I'm happy to review and test once it's there.

cmaglie, 2020-05-28:
@matthijskooijman you may want to have a look here: #720

matthijskooijman, 2020-05-28:
I already had the PR open, waiting for a bit of time to look at it (but you know how it is). Thanks for the heads up, I'll try to review somewhere next week.

closed 2020-06-15 by cmaglie

vifino, 2020-06-15:
Wow, very nice! Thank you @cmaglie!

matthijskooijman, 2020-06-24:
@cmaglie Thanks! I had a look over the code and tried this a little bit today, looking good. Also nice to see that the upload/program/bootloader handling is now all one function, so nice and consistent wrt handling of properties (unlike the Java IDE).

weswitt, 2023-12-10:
Any way to use the avr dragon programmer? I don't see it listed.