arduino/arduino-cli #28: panic: runtime error when running arduino-cli core search arduino

moghaddam, 2018-09-02:
When I try to run the search via core command, I get following error:

**arduino-cli core search arduino**
Searching for platforms matching 'arduino'

panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x8 pc=0x6d93cc]

goroutine 1 [running]:
github.com/arduino/arduino-cli/vendor/go.bug.st/relaxed-semver.(*Version).CompareTo(0x0, 0xc4202dcea0, 0x1)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/go.bug.st/relaxed-semver/version.go:115 +0x2c
github.com/arduino/arduino-cli/vendor/go.bug.st/relaxed-semver.(*Version).GreaterThan(0x0, 0xc4202dcea0, 0x1)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/go.bug.st/relaxed-semver/version.go:202 +0x35
github.com/arduino/arduino-cli/arduino/cores.(*Platform).latestReleaseVersion(0xc420264380, 0xc420279790)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/arduino/cores/cores.go:149 +0x7f
github.com/arduino/arduino-cli/arduino/cores.(*Platform).GetLatestRelease(0xc420264380, 0xc42009b070)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/arduino/cores/cores.go:123 +0x2b
github.com/arduino/arduino-cli/commands/core.runSearchCommand(0xc42026b680, 0xc420255040, 0x1, 0x1)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/commands/core/search.go:59 +0x2dd
github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra.(*Command).execute(0xc42026b680, 0xc420255000, 0x1, 0x1, 0xc42026b680, 0xc420255000)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra/command.go:760 +0x2c1
github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra.(*Command).ExecuteC(0xc42022cc80, 0xc42026b900, 0xc42026a500, 0xc42022db80)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra/command.go:846 +0x30a
github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra.(*Command).Execute(0xc42022cc80, 0x9e50c8, 0xc420215f78)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra/command.go:794 +0x2b
main.main()
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/main.go:30 +0x27

I'm using Ubuntu 18.04
4.15.0-33-generic #36-Ubuntu SMP Wed Aug 15 16:00:05 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

cmaglie, 2018-09-10:
@moghaddam 

May you give some more info about your setup?
Which cores did you install? Maybe you installed some core manually in the `sketchbook/hardware` folder?

moghaddam, 2018-09-10:
No, I don't have the hardware folder on my sketchbook. Also I can't check what core is installed, as I get the same error when I try to list my cores.
But I just noticed in my stack trace (also mentioned in the issue description) that it's referring to the /home/jenkins* while my username is something else and I don't have such user in my system. There is no _jenkins_ in my /etc/passwd file. Is it something hardcoded somewhere?

cmaglie, 2018-09-10:
`jenkins` in the path is normal, is our CI server that makes the build, the path is stored in the executable to give a reference to the source code when a `panic` happens.

May you send me a zipped copy of your `$HOME/.arduino15` folder to c.maglie@arduino.cc?
(it may be several megabytes, you can use wetransfer.com to send it)

moghaddam, 2018-09-10:
OK. I will send it via wetransfer.com. It's getting uploaded.

cmaglie, 2018-09-10:
I've tried with your package but still cannot reproduce the issue, very weird...
Here what I get:

```
$ arduino-cli core search arduino
Searching for platforms matching 'arduino'

ID               	Version	Installed	Name                                             
Intel:arc32      	2.0.2  	No       	Intel Curie Boards                               
arduino:avr      	1.6.21 	No       	Arduino AVR Boards                               
arduino:nrf52    	1.0.2  	No       	Arduino nRF52 Boards                             
arduino:sam      	1.6.11 	No       	Arduino SAM Boards (32-bits ARM Cortex-M3)       
arduino:samd     	1.6.19 	No       	Arduino SAMD Boards (32-bits ARM Cortex-M0+)     
arduino:samd_beta	1.6.24 	No       	Arduino SAMD Beta Boards (32-bits ARM Cortex-M0+)
arduino:stm32f4  	1.0.1  	No       	Arduino STM32F4 Boards                           
esp8266:esp8266  	2.4.2  	Yes      	esp8266                                          
littleBits:avr   	1.0.0  	No       	littleBits Arduino AVR Modules                   

$ arduino-cli core list
ID             	Installed	Latest	Name   
esp32:esp32    	1.0.0    	1.0.0 	esp32  
esp8266:esp8266	2.4.2    	2.4.2 	esp8266
```

let's try in another way, could you run the following command and copy the output here:
1. `arduino-cli config dump`
(I want to be sure that we are using the same 3rd party hardware external URL)
2. `arduino-cli cores search arduino --debug`

moghaddam, 2018-09-11:
Here is my config dump:

```
proxy_type: auto
sketchbook_path: /home/ezm/projects/arduino
arduino_data: /home/ezm/.arduino15
board_manager:
  additional_urls:
  - https://dl.espressif.com/dl/package_esp32_index.json
  - http://arduino.esp8266.com/stable/package_esp8266com_index.json

```

And the output of the `arduino-cli cores search arduino --debug` is:

```
INFO[0000] Initiating configuration
INFO[0000] Unserializing configurations from /home/ezm/applications/arduino-1.8.6/.cli-config.yml
WARN[0000] Error reading config, using default configuration  error="open /home/ezm/applications/arduino-1.8.6/.cli-config.yml: no such file or directory"
WARN[0000] Did not manage to get config file, using default configuration  error="open /home/ezm/applications/arduino-1.8.6/.cli-config.yml: no such file or directory"
INFO[0000] Checking if CLI is Bundled into the IDE
INFO[0000] Candidate IDE Directory: /home/ezm/applications/arduino-1.8.6
INFO[0000] CLI is bundled into the IDE
INFO[0000] Unserializing from IDE preferences
INFO[0000] Configuration set
INFO[0000] Starting root command preparation (`arduino`)
INFO[0000] Formatter set
INFO[0000] Loading the default Package index
INFO[0000] Loading hardware from: /home/ezm/applications/arduino-1.8.6/hardware
INFO[0000] Loading package arduino from: /home/ezm/applications/arduino-1.8.6/hardware/arduino
INFO[0000] Loaded platform                               platform="arduino:avr@"
INFO[0000] Excluding directory: /home/ezm/applications/arduino-1.8.6/hardware/tools
INFO[0000] Loading hardware from: /home/ezm/.arduino15/packages
INFO[0000] Loading package esp32 from: /home/ezm/.arduino15/packages/esp32/hardware
INFO[0000] Loaded platform                               platform="esp32:esp32@1.0.0"
INFO[0000] Checking existence of 'tools' path: /home/ezm/.arduino15/packages/esp32/tools
INFO[0000] Loading tools from dir: /home/ezm/.arduino15/packages/esp32/tools
INFO[0000] Loaded tool                                   tool="esp32:esptool@2.3.1"
INFO[0000] Loaded tool                                   tool="esp32:mkspiffs@0.2.3"
INFO[0000] Loaded tool                                   tool="esp32:xtensa-esp32-elf-gcc@1.22.0-80-g6c4433a-5.2.0"
INFO[0000] Loading package esp8266 from: /home/ezm/.arduino15/packages/esp8266/hardware
INFO[0000] Loaded platform                               platform="esp8266:esp8266@2.4.2"
INFO[0000] Checking existence of 'tools' path: /home/ezm/.arduino15/packages/esp8266/tools
INFO[0000] Loading tools from dir: /home/ezm/.arduino15/packages/esp8266/tools
INFO[0000] Loaded tool                                   tool="esp8266:esptool@0.4.13"
INFO[0000] Loaded tool                                   tool="esp8266:mkspiffs@0.2.0"
INFO[0000] Loaded tool                                   tool="esp8266:xtensa-lx106-elf-gcc@1.20.0-26-gb404fb9-2"
INFO[0000] Loading tools from bundle dir: /home/ezm/applications/arduino-1.8.6/hardware/tools
INFO[0000] Found builtin_tools_versions.txt
INFO[0000] Loaded tool                                   tool="arduino:avrdude@6.3.0-arduino14"
INFO[0000] Loaded tool                                   tool="arduino:arduinoOTA@1.2.1"
INFO[0000] Loaded tool                                   tool="arduino:avr-gcc@5.4.0-atmel3.6.1-arduino2"
Searching for platforms matching 'arduino'

panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x8 pc=0x6d93cc]

goroutine 1 [running]:
github.com/arduino/arduino-cli/vendor/go.bug.st/relaxed-semver.(*Version).CompareTo(0x0, 0xc4202e0120, 0xffffffffffffffff)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/go.bug.st/relaxed-semver/version.go:115 +0x2c
github.com/arduino/arduino-cli/vendor/go.bug.st/relaxed-semver.(*Version).GreaterThan(0x0, 0xc4202e0120, 0x0)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/go.bug.st/relaxed-semver/version.go:202 +0x35
github.com/arduino/arduino-cli/arduino/cores.(*Platform).latestReleaseVersion(0xc420264380, 0xc420350340)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/arduino/cores/cores.go:149 +0x7f
github.com/arduino/arduino-cli/arduino/cores.(*Platform).GetLatestRelease(0xc420264380, 0xc420330570)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/arduino/cores/cores.go:123 +0x2b
github.com/arduino/arduino-cli/commands/core.runSearchCommand(0xc42026b900, 0xc420250800, 0x1, 0x2)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/commands/core/search.go:59 +0x2dd
github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra.(*Command).execute(0xc42026b900, 0xc4202507a0, 0x2, 0x2, 0xc42026b900, 0xc4202507a0)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra/command.go:760 +0x2c1
github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra.(*Command).ExecuteC(0xc42022af00, 0xc42026bb80, 0xc42026a780, 0xc42026a000)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra/command.go:846 +0x30a
github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra.(*Command).Execute(0xc42022af00, 0x9e50c8, 0xc420215f78)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra/command.go:794 +0x2b
main.main()
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/main.go:30 +0x27
```

cmaglie, 2018-09-11:
Ok, now I can reproduce the error, thanks!
When the `arduino-cli` is bundled with the Arduino IDE (i.e. when it is installed in the same folder as the Arduino IDE) the cli reads also the cores bundled in the Java IDE in the `hardware` directory.

Now I'm investigating the reason of the crash...

cmaglie, 2018-09-14:
Ok, it took a while but finally I've pushed a fix in `master` for this:

- if the CLI founds the Arduino IDE it uses also the AVR platform bundled in the IDE' `hardware` folder (for compile, upload, etc.)
- when installing/uninstalling/upgrading cores the bundled platform is ignored, i.e. the CLI operates only in the `ARDUINO_DATA_DIR` (`.arduino15` for Linux).

This actually covers 95% of the cases, there are some corner case that we could handle but it will complicate the package manager logic, for now I'd like to see how this goes.

The fix is available on `master` branch if you build the CLI from source code, or on the next alpha 0.2.2 that I will release shortly.

closed 2018-09-14 by cmaglie

Juraci, 2018-10-03:
I'm having the exact same issue. I'm on Arch Linux and installed using this AUR package https://aur.archlinux.org/packages/arduino-cli-git/

sf750086, 2019-08-02:
hello sir,
I found the same issue but, can't solve it
step no: 5
 commands::=  C:\Users\91940\Documents\Arduino cli>arduino-cli compile --fqbn arduino:av:uno Arduino/test_check
Error: platform not installed
Error during build 

these errors are showing on cmd.  and one more thing next step for 

C:\Users\91940\Documents\Arduino cli>arduino-cli upload -p COM12 --fqbn arduino:avr:uno Arduino/test-check
panic: runtime error: invalid memory address or nil pointer dereference
[signal 0xc0000005 code=0x0 addr=0x0 pc=0x6f49d1]

goroutine 1 [running]:
github.com/arduino/go-paths-helper.(*Path).String(...)
        /home/jenkins/go/pkg/mod/github.com/arduino/go-paths-helper@v0.0.0-20190214132331-c3c98d1bf2e1/paths.go:467
github.com/arduino/arduino-cli/arduino/cores.(*ToolRelease).RuntimeProperties(0x12115bc0, 0x12043dd0)
        /home/jenkins/workspace/arduino-cli-release-builder/arduino/cores/tools.go:138 +0x91
github.com/arduino/arduino-cli/commands/upload.Upload(0xbb15a0, 0x1201e130, 0x1204a500, 0xba9ed0, 0x120060d0, 0xba9ed0, 0x120060d8, 0x12023b60, 0x1f, 0xab33cb)
        /home/jenkins/workspace/arduino-cli-release-builder/commands/upload/upload.go:119 +0x8bf
github.com/arduino/arduino-cli/cli/upload.run(0x1210ba40, 0x12079a10, 0x1, 0x5)
        /home/jenkins/workspace/arduino-cli-release-builder/cli/upload/upload.go:77 +0x146
github.com/spf13/cobra.(*Command).execute(0x1210ba40, 0x120799e0, 0x5, 0x6, 0x1210ba40, 0x120799e0)
        /home/jenkins/go/pkg/mod/github.com/spf13/cobra@v0.0.5/command.go:830 +0x1f1
github.com/spf13/cobra.(*Command).ExecuteC(0x120a7180, 0x976af8, 0x97809c, 0x120241b0)
        /home/jenkins/go/pkg/mod/github.com/spf13/cobra@v0.0.5/command.go:914 +0x252
github.com/spf13/cobra.(*Command).Execute(...)
        /home/jenkins/go/pkg/mod/github.com/spf13/cobra@v0.0.5/command.go:864
main.main()
        /home/jenkins/workspace/arduino-cli-release-builder/main.go:30 +0x20

please suggest me.

sf750086, 2019-08-02:
C:\Users\91940\Documents\Arduino cli> arduino-cli version --debug

INFO[0000] Checking for config file in: C:\Users\91940\AppData\Local\Arduino15\arduino-cli.yaml
INFO[0000] Reading configuration from C:\Users\91940\AppData\Local\Arduino15\arduino-cli.yaml
INFO[0000] Checking if CLI is Bundled into the IDE
INFO[0000] Candidate IDE Directory: C:\Users\91940\Documents\Arduino cli
INFO[0000] CLI is not bundled into the IDE
INFO[0000] Checking for config in: C:\arduino-cli.yaml
INFO[0000] error loading                                
           error="open C:\\arduino-cli.yaml: The system cannot find the file specified."
INFO[0000] Checking for config in: C:\Users\arduino-cli.yaml
INFO[0000] error loading                                
           error="open C:\\Users\\arduino-cli.yaml: The system cannot find the file specified."
INFO[0000] Checking for config in: C:\Users\91940\arduino-cli.yaml
INFO[0000] error loading                                 
           error="open C:\\Users\\91940\\arduino-cli.yaml: The system cannot find the file specified."
INFO[0000] Checking for config in: C:\Users\91940\Documents\arduino-cli.yaml
INFO[0000] error loading                                
           error="open C:\\Users\\91940\\Documents\\arduino-cli.yaml: The system cannot find the file specified."
INFO[0000] Checking for config in: C:\Users\91940\Documents\Arduino cli\arduino-cli.yaml
INFO[0000] error loading                                 
           error="open C:\\Users\\91940\\Documents\\Arduino cli\\arduino-cli.yaml: The system cannot find the file specified."
INFO[0000] Configuration set
INFO[0000] arduino-cli-0.3.7-alpha.preview
INFO[0000] Starting root command preparation (`arduino`)
INFO[0000] Formatter set
arduino-cli Version: 0.3.7-alpha.preview Commit: 69d86a4 BuildDate: 2019-08-02 13:37:28.2069431 +0000 UTC