arduino/arduino-cli #32: Sketch-local libraries directory?

freiser77, 2018-09-05:
Hello,

I create some sketch for my Otto-robot. For this I use the Otto library. I want to use the Otto library only in the Otto project and want to commit it to the git repository.
Is there a possibility for a local library-directory, like libraries or lib inside the sketchbook directory?

I only discovered an libraries folder inside of the Arduino folder which seems to provide the library to all my arduino projects.

Greets 

Florian

brusherru, 2018-09-25:
Yeah. It will be awesome to have a possibility to tell `arduino-cli` where to find out libraries.
It will make work on big projects easier.
And it will be super awesome if I can specify few directories with libraries.

Something like this:
`arduino-cli compile -b arduino:avr:uno -p /dev/tty.usbmodem1144 --libraries="/usr/me/mylibs" mySketch`

Or just specify library directories in the `.cli-config.yml`:
```
libraries:
- /usr/me/my_libs
- /usr/me/some_project/library
```

masci, 2019-08-05:
Let's consolidate the discussion here: https://github.com/arduino/arduino-cli/issues/108

closed 2019-08-05 by masci