arduino/arduino-cli #33: Compilation error for Arduino/Genuino 101

ricms93, 2018-09-06:
Hi,
I am compiling an Arduino Code I have for an Arduino 101 embedded into a UDOO x86 board. After running the compilation command with the correct fqbn I get this error:

Sketch uses 49356 bytes (31%) of program storage space. Maximum is 155648 bytes.
Error: open ../arduino_files/serial_main_push/serial_main_push.ino/serial_main_push.ino.Intel.arc32.arduino_101.hex: not a directory
Error copying output file.

Therefore I cannot continue uploading the compiled file into the board. Is there something I'm missing? 
Thanks in advance.

My OS is Ubuntu 16.04 LTS

wjwieland, 2018-10-23:
I get the same error.  Where does arduino-cli put compiled files when done?  And can that be dictated on the cli or via a config file?

wjwieland, 2018-10-26:
I found out what is going on here, at least on my RPi 3.  The command for compile should NOT include the filename of the .ino file.  It should only point to the directory/folder that the .ino file resides in.

for example:  I had been entering the following:
` arduino-cli compile -b arduino:avr:mega ~/Devel/threeaxis/treeaxis.ino`
which resulted in the compile error we both have been seeing.  When, however, I enter the following:
`arduino-cli compile -b arduino:avr:mega ~/Devel/threeaxis/`
All works as expected and I end up  with the following in the folder noted in the second command string:

wjw@rpi-home bin $ ll ~/Devel/threeaxis/
```
total 96
drwxr-xr-x 2 wjw wjw  4096 Oct 26 11:43 ./
drwxr-xr-x 5 wjw wjw  4096 Oct 25 00:36 ../
-rwxr-xr-x 1 wjw wjw 58348 Oct 26 11:43 threeaxis.arduino.avr.mega.elf*
-rw-r--r-- 1 wjw wjw 21290 Oct 26 11:43 threeaxis.arduino.avr.mega.hex
-rw-r--r-- 1 wjw wjw  2335 Oct 25 00:42 threeaxis.ino
```

Now to get upload to work...which I have not yet.

lmiller1990, 2019-04-24:
The above helped me too - you need to specify the directory of the `.ino` file, not the `.ino` file. After doing so, mine compiled. Then I was able to upload it by running:

```
$ a board list
FQBN    Port                   	ID       	Board Name
    	/dev/cu.usbserial-14110	1A86:7523	unknown
```

To get the port, then

```
a upload -p /dev/cu.usbserial-14110 --fqbn arduino:avr:uno MyFirstSketch
```

Where `a` is aliased to the arduino-cli.

closed 2019-05-24 by per1234

per1234, 2019-05-24:
Closing as resolved. We have a separate issue requesting that this situation either be handled automatically or that the documentation mention that only the sketch folder should be specified in the command: https://github.com/arduino/arduino-cli/issues/51