arduino/arduino-cli #34: GPIO-based DTR reset needed for upload on Raspberry Pi/UP2 board etc.

deadprogram, 2018-09-07:
First of all, thanks for work on this project, I have been working a Golang-based CLI for the Arduino toolchain for the longest time.

It would appear that in order to upload code to an Arduino that is connected directly to the `ttyAMA0` serial port via the pins on a Raspberry Pi, UP2 board or other board, a GPIO pin-based DTR reset needs to be implemented.

Such like an implementation could be an additional flag named something like `gpioreset` that is conditional similar to what https://github.com/arduino/arduino-cli/blob/master/commands/upload/upload.go#L212 does, but that pulls the GPIO pin high/low similar to https://github.com/platformio/platformio-core/pull/687/files#diff-9df12ed969b3177c694a1020f4de5cc8R82 but of course allowing a specific GPIO pin to be specified.

For example:

```
arduino-cli upload -p /dev/ttyAMA0 -gpioreset 4 --fqbn atmel-avr-xminis:avr:atmega328 Arduino/MyFirstSketch
```

would use GPIO4 to perform the reset before attempting the upload.

Any thoughts on this?

deadprogram, 2018-09-25:
Anyone have any opinion on this?

deadprogram, 2018-09-26:
Hi @cmaglie I have just gone thru the CLI code, and realized that the CLI is literally just a CLI and still needs all of the build tools for any give platform such as avrdude to perform the upload. I had not dug far enough into the CLI code to realize that it is not a replacement for the toolchain.

So there are 2 techniques used in order to deal with the DTR being a separate pin when programming an Arduino from for example a Raspberry Pi or UP2 board.

- a patched version of avrdude: ala http://www.kfu.com/~nsayer/geppetto/linuxgpio-patch.txt
- a Linux system hack using strace combined with some python ala https://github.com/deanmao/avrdude-rpi

It seems useful to be able to do the ioctrl strace hack for the user based on command flags passed into the `arduino-cli upload` command. Alternative would be I suppose use the cli to install the tools, and then manually do the symlinking based on the install location used by the CLI.

deadprogram, 2018-09-26:
That looks very cool @facchinm thanks for the reply.

The board I am wanting to flash is the GrovePi. This is the script that does it using a patched avrdude: https://github.com/RobertLucian/GrovePi/blob/master/Firmware/grovepi_firmware_update.sh

Pinging @RobertLucian will this allow replacing the custom patched avrdude you are maintaining for @DexterInd for the mainline version?

RobertLucian, 2018-10-08:
@deadprogram I'm gonna have a look at this stuff - it's been a while since we looked into `avrdude`. I think it's very likely that this can replace ours. I also love the idea of having mainstream support to it, which is great!

weswitt, 2018-12-10:
I flash Arduino boards from an RPI all the time using the linuxgpio programmer. Just connect your pi to the Arduino using SPI and it works great.

masci, 2019-08-05:
Closing since discussion has been stale for a while, feel free to re-open if latest messages didn't work.

closed 2019-08-05 by masci

dwright213, 2019-11-20:
I'm still running into this `Error during Upload: cannot perform reset: cannot set DTR` message when I attempt to upload. 

The board I'm using is an Arduino zero compatible, Mechaduino board. Is there some additional piece of software am going to need to upload to this board?

noeldum, 2022-02-24:
Nearly 4 years later what is the conclusion on this?

How do I control a DTR line from arduino-cli? Has the suggested solution been implemented?

For portability is it possible to have arduino-cli triggering a reset command rather than gpio direct control maybe? Then the user would write this reset command.

@deadprogram how did you sort this in the end?

PR #345 (2019-08-28): Rewrite the output system
This PR rewrites the whole logging and feedback system of the CLI. It also fixes #34 

# Rationale

This PR implements a blueprint where the logging system is detached from the output produced in response to user actions.

## Logging

The logging system is used by low level components to leave marks at any step of the code execution and it's intended to be mainly consumed by machines. Structured logging will be encouraged and a review of what we're logging right now, the logging levels and the structure of the messages will be eventually needed. To avoid passing aroun [... cut]