arduino/arduino-cli #35: Compiling for ESP8285 fails: core/core.a: no such file or directory

laleporo, 2018-09-07:
Hi there,

I am trying to compile a sketch (any sketch can reproduce, even a basic one without any code) for the ESP8285 on an Ubuntu 16.04.5 LTS system. After downloading the precompiled binary 0.2.1-alpha.preview for 64bit, adding the board manager in the ".cli-config.yml"
```
board_manager:
  additional_urls:
    - http://arduino.esp8266.com/stable/package_esp8266com_index.json
```
and installing the esp arduino core (version 2.4.2)
```
arduino-cli core install esp8266:esp8266
```
the build process can be started using the default build properties from the IDE:
```
arduino-cli compile --fqbn esp8266:esp8266:esp8285 --build-properties="build.f_cpu=80000000L,build.vtable_flags=-DVTABLES_IN_FLASH,upload.resetmethod=none,build.flash_size=1M,build.flash_size_bytes=0x100000,build.flash_ld=eagle.flash.1m0.ld,build.spiffs_pagesize=256,upload.maximum_size=1023984,build.rfcal_addr=0xFC000,build.led=-DLED_BUILTIN=0,build.lwip_include=lwip2/include,build.lwip_lib=-llwip2,build.lwip_flags=-DLWIP_OPEN_SRC -DTCP_MSS=536,build.debug_port=,build.debug_level=,upload.erase_cmd=,upload.speed=115200"
```

On a windows system with an installed IDE I end up with a built sketch. On the linux system however, there is an error:
```
Error: open /tmp/arduino-sketch-9A982EFD8A9D637ECC6D29F90E4C1064/core/core.a: no such file or directory
Compilation failed.
```

The file core/core.a is not generated on both systems. Additional build logs can be supplied if needed.

cjsieh, 2018-09-07:
I patched esp8266-2.4.2/platform.txt to change the location in the recipe.ar pattern and recipe.ar.combine.pattern  .  I got this location from looking at the "avr" recipe.ar.pattern .  

<pre>
-- esp8266-2.4.2/platform.txt	2018-08-03 02:19:33.000000000 -0500
+++ platform.txt	2018-09-07 11:17:45.808640177 -0500
@@ -92,10 +92,10 @@
 recipe.S.o.pattern="{compiler.path}{compiler.c.cmd}" {compiler.cpreprocessor.flags} {compiler.S.flags} -DF_CPU={build.f_cpu} {build.lwip_flags} {build.debug_port} {build.debug_level} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} -DARDUINO_BOARD="{build.board}" {build.led} {compiler.c.extra_flags} {build.extra_flags} {includes} "{source_file}" -o "{object_file}"
 
 ## Create archives
-recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/arduino.ar" "{object_file}"
+recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}"
 
 ## Combine gc-sections, archives, and objects
-recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" -Wl,-Map "-Wl,{build.path}/{build.project_name}.map" {compiler.c.elf.flags} {compiler.c.elf.extra_flags} -o "{build.path}/{build.project_name}.elf" -Wl,--start-group {object_files} "{build.path}/arduino.ar" {compiler.c.elf.libs} -Wl,--end-group  "-L{build.path}"
+recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" -Wl,-Map "-Wl,{build.path}/{build.project_name}.map" {compiler.c.elf.flags} {compiler.c.elf.extra_flags} -o "{build.path}/{build.project_name}.elf" -Wl,--start-group {object_files} "{build.path}/{archive_file}" {compiler.c.elf.libs} -Wl,--end-group  "-L{build.path}"
 
 ## Create eeprom
 recipe.objcopy.eep.pattern=
</pre>

laleporo, 2018-09-07:
The provided patch works.
So I guess it's not a problem with arduino-cli but with the ESP core.

closed 2018-09-07 by laleporo

cjsieh, 2018-09-07:
The arduino developers said they only tested "arduino" cores.  ESP32 is also broken. So more work needs to still be done.

cjsieh, 2018-09-10:
This is only my solution to this issue.  Arduino developers may have a "better" solution.