arduino/arduino-cli #36: Arduino-cli breaks Arduino-ide and Vice Versa

NormanDunbar, 2018-09-08:
While experimenting with the cli I noticed that when I next open the IDE (Arduino 1.8.5) nothing happens after the splash screen. If I open it again, then the IDE works fine. Sketches upload etc. It's just the first time after using the cli.

However, after using the IDE, the CLI no longer seems to know where the `package_index.json` is to be found, and it has to be downloaded again. The error is:

````
$ arduino-cli board list

Error: loading json index file /home/norman/.arduino15/package_index.json: open /home/norman/.arduino15/package_index.json: no such file or directory
Failed to load https://downloads.arduino.cc/packages/package_index.json package index.
Try updating all indexes with `arduino-cli core update-index`.
````

a7hybnj2, 2019-02-12:
I have this same issue... timemage on irc lead me to find
```
./arduino
cc.arduino.contributions.SignatureVerificationFailedException: package_index.json file signature verification failed
	at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:91)
	at processing.app.BaseNoGui.initPackages(BaseNoGui.java:534)
	at processing.app.Base.<init>(Base.java:270)
	at processing.app.Base.main(Base.java:153)
```

this is the error I get when running arduino from cli after using arduino-cli. I don't know what it means but I hope it helps.

MacOS

a7hybnj2, 2019-02-12:
I don't exactly know how the issue first started but it was caused by not having *.json.sig files after using arduino-cli. I was able to fix it for now by opening IDE - board manager. This populated the *.json.sig files and now I can switch between cli and IDE without issue (for now).

There were some things I did in between like deleting the index files but I don't think any of that was really needed.

NormanDunbar, 2019-02-13:
This problem was on the early alphas, it might be fixed - I'll test ASAP.

One other thing I noticed, after installing the cli and running it, my IDE now picks up the files on /home/me/.arduino15brather than where I extracted the original IDE zip file.

Linux Mint 19.1.

Cheers,
Norm.

a7hybnj2, 2019-02-14:
@NormanDunbar I had this issue with the latest alpha and Arduino IDE beta 1.9 on MacOS 10.14.3

NormanDunbar, 2019-03-23:
Ok, I have finally downloaded the latest Linux cli, version 0.3.6, and tested it. Good news, my original issue appears fixed. I can swap between the CLI and the IDE at will. I'm happy that this is now fixed.

However, I now can confirm that whenever I've installed the CLI and run at least one compilation - although it might be just installing a platform - the IDE swaps from using the files in its own folders to using the ones in the `~/.arduino15`, which actually belong to the CLI and not the IDE.

Tested with a clean install of IDE 1.8.5 (which I can't yet upgrade) and CLI 0.3.6.

I shall raise this as a separate issue as this issue is a different matter and is now fixed.

Cheers,
Norm.

closed 2019-03-23 by NormanDunbar