arduino/arduino-cli #37: Arduino-cli cannot find Duemilanove or Uno

NormanDunbar, 2018-09-08:
I have a genuino Arduino Deumilanove and an Uno clone with the ch341 chip and a genuine Mega 2560 R3, none of which are detected by the arduino-cli and cannot be used.

Dell Vostro 1720 64bit
Linux Mint 18.3
Kernel 4.13.0-45

My user is in the `dialout` group.

I installed the Linux 64bit preview version of Arduino-cli and proceeded as follows:

````
$ arduino-cli core update-index
Updating index: package_index.json downloaded
````

Then:

````
norman@Hubble ~ $ arduino-cli board list
FQBN    Port        	ID       	Board Name
    	/dev/ttyUSB0	0403:6001	unknown   
````
Search for core:

````
norman@Hubble ~ $ arduino-cli core search arduino
Searching for platforms matching 'arduino'

ID               	Version	Installed	Name                                             
Intel:arc32      	2.0.2  	No       	Intel Curie Boards                               
arduino:avr      	1.6.21 	No       	Arduino AVR Boards                               
arduino:nrf52    	1.0.2  	No       	Arduino nRF52 Boards                             
arduino:sam      	1.6.11 	No       	Arduino SAM Boards (32-bits ARM Cortex-M3)       
arduino:samd     	1.6.19 	No       	Arduino SAMD Boards (32-bits ARM Cortex-M0+)     
arduino:samd_beta	1.6.24 	No       	Arduino SAMD Beta Boards (32-bits ARM Cortex-M0+)
arduino:stm32f4  	1.0.1  	No       	Arduino STM32F4 Boards                           
littleBits:avr   	1.0.0  	No       	littleBits Arduino AVR Modules                   
````

Install arduino:avr core

````
norman@Hubble ~ $ arduino-cli core install arduino:avr

Downloading arduino:avr-gcc@4.9.2-atmel3.5.4-arduino2...
arduino:avr-gcc@4.9.2-atmel3.5.4-arduino2 downloaded                                                                    
Downloading arduino:avrdude@6.3.0-arduino9...
arduino:avrdude@6.3.0-arduino9 downloaded                                                                               
Downloading arduino:arduinoOTA@1.1.1...
arduino:arduinoOTA@1.1.1 downloaded                                                                                     
Downloading arduino:avr@1.6.21...
arduino:avr@1.6.21 downloaded                                                                                           
Installing arduino:avr-gcc@4.9.2-atmel3.5.4-arduino2
arduino:avr-gcc@4.9.2-atmel3.5.4-arduino2 installed
Installing arduino:avrdude@6.3.0-arduino9
arduino:avrdude@6.3.0-arduino9 installed
Installing arduino:arduinoOTA@1.1.1
arduino:arduinoOTA@1.1.1 installed
Installing arduino:avr@1.6.21...
arduino:avr@1.6.21 installed
````

Check core:

````
norman@Hubble ~ $ arduino-cli core list

ID         	Installed	Latest	Name              
arduino:avr	1.6.21   	1.6.21	Arduino AVR Boards
````

Check board again:

````
norman@Hubble ~ $ arduino-cli board list

FQBN    Port        	ID       	Board Name
    	/dev/ttyUSB0	0403:6001	unknown   
````

Seems my Duemilanove is not recognised. Tried with the Uno same result. The Mega on the other hand is found:

````
$ arduino-cli board list

FQBN            	Port        	ID       	Board Name                       
arduino:avr:mega	/dev/ttyACM0	2341:0042	Arduino/Genuino Mega or Mega 2560
````

Cheers,
Norm.

NormanDunbar, 2018-09-11:
### UPDATE:

I can now compile and upload sketches to my Uno and Duemilanove with no problems even though the boards themselves are not recognised.

#### Uno

````
arduino-cli compile --fqbn arduino:avr:uno MyFirstSketch
arduino-cli upload -p /dev/ttyUSB0 --fqbn arduino:avr:uno MyFirstSketch
````

#### Duemilanove

````
arduino-cli compile --fqbn arduino:avr:diecimila:cpu=atmega328 MyFirstSketch
arduino-cli upload -p /dev/ttyUSB0 --fqbn arduino:avr:diecimila:cpu=atmega328 MyFirstSketch
````

This might help someone. In the meantime though, the original problem of the boards not being recognised remains.

Cheers,
Norm.

cmaglie, 2018-09-20:
could you paste the output of `lsusb` with and without the board connected?

NormanDunbar, 2018-09-21:
Morning @cmaglie   As requested, here is the output from `lsusb` when the devices are plugged in. I've also included output from `dmesg` as listed when each device was initially plugged in.

HTH

Cheers,
Norm.

## Nothing Plugged In Yet

````
norman@Hubble ~ $ lsusb
Bus 002 Device 003: ID 0c45:63e0 Microdia Sonix Integrated Webcam
Bus 002 Device 002: ID 1058:07a8 Western Digital Technologies, Inc. My Passport (WDBBEP), My Passport for Mac (WDBLUZ)
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 008 Device 005: ID 0a5c:4503 Broadcom Corp. Mouse (Boot Interface Subclass)
Bus 008 Device 004: ID 0a5c:4502 Broadcom Corp. Keyboard (Boot Interface Subclass)
Bus 008 Device 003: ID 413c:8126 Dell Computer Corp. Wireless 355 Bluetooth
Bus 008 Device 002: ID 0a5c:4500 Broadcom Corp. BCM2046B1 USB 2.0 Hub (part of BCM2046 Bluetooth)
Bus 008 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 002: ID 062a:4102 Creative Labs 
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
````

## Plug in Duemilanove

### Dmesg Output

````
[  266.620317] usb 7-1: new full-speed USB device number 2 using uhci_hcd
[  266.830631] usb 7-1: New USB device found, idVendor=0403, idProduct=6001
[  266.830640] usb 7-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  266.830646] usb 7-1: Product: FT232R USB UART
[  266.830651] usb 7-1: Manufacturer: FTDI
[  266.830656] usb 7-1: SerialNumber: A700dXsq
[  268.387831] usbcore: registered new interface driver usbserial
[  268.387855] usbcore: registered new interface driver usbserial_generic
[  268.387872] usbserial: USB Serial support registered for generic
[  268.394095] usbcore: registered new interface driver ftdi_sio
[  268.394213] usbserial: USB Serial support registered for FTDI USB Serial Device
[  268.394412] ftdi_sio 7-1:1.0: FTDI USB Serial Device converter detected
[  268.395211] usb 7-1: Detected FT232RL
[  268.397633] usb 7-1: FTDI USB Serial Device converter now attached to ttyUSB0
````

### Lsusb Output

````
norman@Hubble ~ $ lsusb
Bus 002 Device 003: ID 0c45:63e0 Microdia Sonix Integrated Webcam
Bus 002 Device 002: ID 1058:07a8 Western Digital Technologies, Inc. My Passport (WDBBEP), My Passport for Mac (WDBLUZ)
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 008 Device 005: ID 0a5c:4503 Broadcom Corp. Mouse (Boot Interface Subclass)
Bus 008 Device 004: ID 0a5c:4502 Broadcom Corp. Keyboard (Boot Interface Subclass)
Bus 008 Device 003: ID 413c:8126 Dell Computer Corp. Wireless 355 Bluetooth
Bus 008 Device 002: ID 0a5c:4500 Broadcom Corp. BCM2046B1 USB 2.0 Hub (part of BCM2046 Bluetooth)
Bus 008 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 007 Device 002: ID 0403:6001 Future Technology Devices International, Ltd FT232 USB-Serial (UART) IC
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 002: ID 062a:4102 Creative Labs 
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
````

## Plug in Uno (Clone)

### Dmesg Output

````
[  394.728408] usb 7-1: new full-speed USB device number 3 using uhci_hcd
[  394.906390] usb 7-1: New USB device found, idVendor=1a86, idProduct=7523
[  394.906399] usb 7-1: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[  394.906404] usb 7-1: Product: USB2.0-Serial
[  396.499976] usbcore: registered new interface driver ch341
[  396.500273] usbserial: USB Serial support registered for ch341-uart
[  396.500330] ch341 7-1:1.0: ch341-uart converter detected
[  396.506257] usb 7-1: ch341-uart converter now attached to ttyUSB0
````

### Lsusb Output

````
Bus 002 Device 003: ID 0c45:63e0 Microdia Sonix Integrated Webcam
Bus 002 Device 002: ID 1058:07a8 Western Digital Technologies, Inc. My Passport (WDBBEP), My Passport for Mac (WDBLUZ)
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 008 Device 005: ID 0a5c:4503 Broadcom Corp. Mouse (Boot Interface Subclass)
Bus 008 Device 004: ID 0a5c:4502 Broadcom Corp. Keyboard (Boot Interface Subclass)
Bus 008 Device 003: ID 413c:8126 Dell Computer Corp. Wireless 355 Bluetooth
Bus 008 Device 002: ID 0a5c:4500 Broadcom Corp. BCM2046B1 USB 2.0 Hub (part of BCM2046 Bluetooth)
Bus 008 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 007 Device 003: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 002: ID 062a:4102 Creative Labs 
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
````

PaulStoffregen, 2018-09-25:
> In the meantime though, the original problem of the boards not being recognised remains.

Isn't "unknown" the expected behavior for a CH340-based clone?

cmaglie, 2018-09-26:
> Isn't "unknown" the expected behavior for a CH340-based clone?

Yes, @NormanDunbar your boards are cheap clones that mounts the CH340 USB2Serial converter. There is no problem with that, except that it provides a generic USB VID/PID `1a86:7523` that cannot be associated with a specific board.

I'll close this as invalid. If I'm missing something please comment further and it will be eventually reopened.

closed 2018-09-26 by cmaglie

NormanDunbar, 2018-09-26:
Morning @cmaglie  @PaulStoffregen.

I accept that the Uno is a clone, I believe I mentioned it and yes, it is based on the CH340. However, the Duemilanove is genuine and isn't, it's  not CH340, it's FTDI.

Cheers,
Norm.

NormanDunbar, 2018-09-30:
Morning @cmaglie @PaulStoffregen,

A further comment. My Mega is also a clone, as it turns out, and that is recognised, apparently as a genuine Mega 2560. See above.

Cheers,
Norm.

cmaglie, 2018-10-03:
@NormanDunbar it depends on the USB VID/PID that the "board" reports to the OS.

Some very old boards like the Duemilanove or some cheap clones, uses an FTDI-like chip to do UART-to-USB conversion. This chip always reports the same VID/PID so all boards using an FTDI or a CH341 will always report the same VID/PID: it's not possible to detect which board model mounts that chip.

You can check the VID/PID looking at the `lsusb` output:

CH340 will have `1a86:7523`

```
Bus 007 Device 003: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter
```

FT232 have `0403:6001`

```
Bus 007 Device 002: ID 0403:6001 Future Technology Devices International, Ltd FT232 USB-Serial (UART) IC
```

you will find that those IDs are shared across all boards using the same USB-2-UART chip, so the only thing you can say is that the board is using an FTDI for example, but you can't tell which board is exactly.

nickiesiva, 2019-03-10:
> I can now compile and upload sketches to my Uno and Duemilanove with no problems even though the boards themselves are not recognised.
> #### Uno
> 
> ```
> arduino-cli compile --fqbn arduino:avr:uno MyFirstSketch
> arduino-cli upload -p /dev/ttyUSB0 --fqbn arduino:avr:uno MyFirstSketch
>```

I can compile, but can't upload. I got this errors :
```
Error: platform arduino:avr is not installed
Invalid FQBN.
```

`arduino-cli core list` :
```
ID                      Installed       Latest  Name              
arduino:avr@1.6.23      1.6.23          1.6.23  Arduino AVR Boards

```

Can anyone explain,  why I got that errors ?