arduino/arduino-cli #41: Does not detecting Arduino nano

stechiez, 2018-09-15:
Hi
i was trying to connect Arduino nano with RPI. RPI was not able to detect the Arduino nano.
I installed the core for Arduino:avr, arduino:samd
but then also not able to discover. i reset the controller manually as well but no effect..

tab4moji, 2018-09-17:
genuine product or clone ?

i have nano clones.
my rpi failed to detect clones, but rpi could compile and upload my code completely.

stechiez, 2018-09-17:
I have Arduino nano clone from aliexpress. 
Could you please tell us, how did you succeeded with flashing.
i couldn't able to discover the board.

tab4moji, 2018-09-17:
will #29 or #38 help you?

cmaglie, 2018-10-11:
Probably the board can't be discovered, please read:
* https://github.com/arduino/arduino-cli/#why-the-arduino-unomegaduemilanove-is-not-detected-when-i-run-arduino-cli-board-list
* https://github.com/arduino/arduino-cli/issues/37#issuecomment-426697011

BTW even if it's not detected you can compile/upload anyway using the FQBN `arduino:avr:nano` or if you have the older bootloader (and since the board is a clone I suppose it is the older) `arduino:avr:nano:cpu=atmega328old`.

Closing as duplicate of #37, please comment further if this doesn't solve your issue.

closed 2018-10-11 by cmaglie