arduino/arduino-cli #42: Do we have support for ESP8266

embhobbb, 2018-09-15:
Hi,
i have installed CLI interface and able to install the core package for AVR. But i am using ESP8266 core. 
I searched for the core but didn't find. 
`./arduino-cli-0.2.1-alpha.preview-linuxarm core search esp8266
Searching for platforms matching 'esp8266'

No platforms matching your search`

embhobbb, 2018-09-22:
thanks.
i got error while downloading package for esp8266.

```
./arduino-cli core install esp8266:esp8266
Downloading esp8266:esptool@0.4.13...
esp8266:esptool@0.4.13 downloaded
Error: server returned a non-2XX status code
Error downloading esp8266:esptool@0.4.13

```

ljcucc, 2018-09-24:
Same problem in https://github.com/arduino/arduino-cli/issues/43

cmaglie, 2018-09-24:
Duplicate of #43

brusherru, 2018-09-25:
@cmaglie @ljcucc , it's not the same problem in the topic. 
The topic is about adding some instruction about `additional_urls` into `README.md`.

But then author said about another problem, that is reported in #43. ;)
So I propose to reopen this issue and then just add some documentation.

sashad, 2018-11-27:
ls ./.arduino15/
package_esp8266com_index.json  package_index.json
but
arduino-cli core install esp8266:esp8266
Error: package esp8266 not found
Could not determine platform dependencies

zoobab, 2019-01-25:
@brusherru it is not clear where this file ".cli-config.yml" needs to be located. In ~/.cli-config.yml ? In the current directory? I have added it in ~/.cli-config.yml , but it does not find the esp8266 repo, so I guess it is the wrong location...

zoobab, 2019-01-25:
After some debugging, where I copied the arduino-cli binary in /usr/local/bin/, it looks for the ".cli-config.yml" in this directory, so I copied it there, and now it works, weird: 

```
$ arduino-cli core update-index --debug
INFO[0000] Initiating configuration
INFO[0000] Unserializing configurations from /usr/local/bin/.cli-config.yml
WARN[0000] Error reading config, using default configuration  error="open /usr/local/bin/.cli-config.yml: no such file or directory"
WARN[0000] Did not manage to get config file, using default configuration  error="open /usr/local/bin/.cli-config.yml: no such file or directory"
```

per1234, 2019-05-24:
Closing as fixed by https://github.com/arduino/arduino-cli/commit/3dcd1ca4ae071acb9fa00b93bb4e920606ce63cd

closed 2019-05-24 by per1234

adi-g15, 2022-11-30:
> After some debugging, where I copied the arduino-cli binary in /usr/local/bin/, it looks for the ".cli-config.yml" in this directory, so I copied it there, and now it works, weird:
>
> ```
> $ [...]

To add to this, if you don't want to put a config file in root (like if you are on NixOS), you can use the `--config-file` flag to specify the path to `.cli-config.yml`.