arduino/arduino-cli #43: BUG: Сan't install tools if installation was interrupted

brusherru, 2018-09-17:
I catch it when installing `esp8266:esp8266`, so I'll write steps about it.

## Steps to reproduce
1. Add the additional URL into `.cli-config.yml`:
    ```yml
board_manager:
  additional_urls:
    - http://arduino.esp8266.com/stable/package_esp8266com_index.json
    ```
2. Run `arduino-cli core install esp8266:esp8266`
3. It runs installation and shows progress bars...
4. Press "Ctrl+C" (abort installation) while it loads one of tools.
5. Run `arduino-cli core install esp8266:esp8266` again

## Expected behavior
It tries to download tools again and then installs the core.

## Actual behavior
It exits with a message on each run:
```
Downloading esp8266:esptool@0.4.13...
esp8266:esptool@0.4.13 downloaded                                                                
Error: server returned a non-2XX status code
Error downloading esp8266:esptool@0.4.13
```

## How to reproduce again
1. Go to the data folder (by default it is `arduino15` folder)
2. Open `staging/packages` directory and remove all archives with esp tools.
3. Run `arduino-cli core install esp8266:esp8266` — it will download tools again without error about "non-2XX status code".

So... looks like it can't override archive in the staging folder and throws an incorrect error 🤔

cmaglie, 2018-09-20:
Interesting, right now I think that the installer just stops in the middle leaving in an dirty state.
I guess we should catch SIGINT and act accordingly during installation.

ljcucc, 2018-09-24:
I have same problem 2. 😕

*On windows 10:*
```
username@DESKTOP C:\Users\username\bin\Arduino
$ arduino-cli core install esp8266:esp8266

Downloading esp8266:esptool@0.4.13...
esp8266:esptool@0.4.13 downloaded
Error: server returned a non-2XX status code
Error downloading esp8266:esptool@0.4.13
```

Then the **.cli-config.yml** file
``` yml
board_manager: 
  additional_urls:
    - http://arduino.esp8266.com/stable/package_esp8266com_index.json 
```

PR #89 (2018-11-09): Installer improvements
- fixed precodintion check for already downloaded files
- cleanup temps if installation is interrupted with CTRL+C during archive extraction

Should fix #43
closed 2018-11-09 by cmaglie