arduino/arduino-cli #44: Problem with FQBN handling?

vicatcu, 2018-09-19:
I'm using Ubuntu 16.04 x64. I have successfully installed by custom boards manager package by specifying a `.cli-config.yml` that looks like this based on reading other issues here:
```yml
board_manager:
  additional_urls:
    - http://update.wickeddevice.com/package_airqualityegg-2.3.3_index.json
```

Then I do `arduino-cli core update-index` and it gets my boards package no problem.
Then I do `arduino-cli core list` and it outputs the following:
```
ID                             	Installed	Latest	Name           
Air Quality Egg v2 - 2.3.3:avr 	2.3.3    	2.3.3 	Air Quality Egg
```

Now I'm trying to figure out how to compile a sketch using that package and I'm getting tripped up on the FQBN business. If I compile a sketch in the IDE and look at the build.options.json that gets generated in the build folder, I see this:
```json
{ "fqbn": "Air Quality Egg v2 - 2.3.3:avr:wildfiredo3" }
```

But if I try to compile using the following command:

```
arduino-cli compile --fqbn "Air Quality Egg v2 - 2.3.3:avr:wildfiredo3" sketchbook/testing
```
I get an error message which makes no sense to me:

```
Error: Error resolving FQBN: missing package arduino:Air Quality Egg v2 - 2.3.3:avr required for build
Compilation failed.
```

Maybe this has something to do with the structure of my package or naming conventions that needs to be corrected? But the fact remains, I have no problems using my package with the IDE, so I figure it must be a problem with the processing of the fbqn command line argument for the arduino-cli. Help / guidance appreciated.

ignorant, 2018-09-20:
If you don't have arduino:avr core installed,try install arduino:avr core.
After install core,compile again.

cmaglie, 2018-09-20:
Probably the `name` field in the package json http://update.wickeddevice.com/package_airqualityegg-2.3.3_index.json should also be different:

```
{
  "packages": [ {
    "name":"Air Quality Egg v2 - 2.3.3",
    "maintainer":"Wicked Device LLC",
    "websiteURL":"http://airqualityegg.com",
```

probably you want something like:

`"name" : "airqualityegg"`

or 

`"name": "wicked"`

anyway this field is taken as the first part of the FQBN and should be lowercase letters only.
Maybe a check on this field should be added to produce a warning.