arduino/arduino-cli #47: ESP8266 not Compiling

embhobbb, 2018-09-23:
Hi
i have install the package for ESP8266. But i am getting error while compiling.
```
$arduino-cli compile --fqbn esp8266:esp8266:nodemcu:CpuFrequency=80 Arduino/blinking
Error: open /tmp/arduino-sketch-1B85C9E93D6B4FFB66DC17038C29FA1A/core/core.a: no such file or directory
Compilation failed.

```

i referred the the following link as well
#35 

i tried the switched used in the #35 but still no success.

ignorant, 2018-09-26:
#35 
>I patched esp8266-2.4.2/platform.txt to change the location in the recipe.ar pattern and recipe.ar.combine.pattern.

See [Arduino IDE 1.5 3rd party Hardware specification](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification#recipes-to-build-the-corea-archive-file).
{archive_file_path} ; This property was added arduino builder 1.0.0-beta12 as a replacement for {build.path}/{archive_file}.

Try thiis patch.

```
--- platform.txt.org	2018-09-26 17:53:23.098072100 +0900
+++ platform.txt	2018-09-26 18:21:12.893480100 +0900
@@ -92,10 +92,10 @@
 recipe.S.o.pattern="{compiler.path}{compiler.c.cmd}" {compiler.cpreprocessor.flags} {compiler.S.flags} -DF_CPU={build.f_cpu} {build.lwip_flags} {build.debug_port} {build.debug_level} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} -DARDUINO_BOARD="{build.board}" {build.led} {compiler.c.extra_flags} {build.extra_flags} {includes} "{source_file}" -o "{object_file}"
 
 ## Create archives
-recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/arduino.ar" "{object_file}"
+recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}"
 
 ## Combine gc-sections, archives, and objects
-recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" -Wl,-Map "-Wl,{build.path}/{build.project_name}.map" {compiler.c.elf.flags} {compiler.c.elf.extra_flags} -o "{build.path}/{build.project_name}.elf" -Wl,--start-group {object_files} "{build.path}/arduino.ar" {compiler.c.elf.libs} -Wl,--end-group  "-L{build.path}"
+recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" -Wl,-Map "-Wl,{build.path}/{build.project_name}.map" {compiler.c.elf.flags} {compiler.c.elf.extra_flags} -o "{build.path}/{build.project_name}.elf" -Wl,--start-group {object_files} "{archive_file_path}" {compiler.c.elf.libs} -Wl,--end-group  "-L{build.path}"
 
 ## Create eeprom
 recipe.objcopy.eep.pattern=
```

embhobbb, 2018-10-06:
Thanks. I could compile the code with provided patch.

closed 2018-10-06 by embhobbb