arduino/arduino-cli #52: Baud rate forced to 115200

Prior99, 2018-10-02:
When trying to flash my Arduino Nano using this CLI, the CLI forces the Baud rate to `115200`.

I invoked the CLI like this:

```
arduino-cli --debug upload -b arduino:avr:nano:cpu=atmega328 -p /dev/ttyUSB0 -t -v .
```

With this high Baud rate, the board can't get in sync with the CLI.

Using `ps auxw`, I was able to determine how `arduino-cli` calls `avrdude`:

```
/home/prior/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude -C/home/prior/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino9/etc/avrdude.conf -v -patmega328p -carduino -P/dev/ttyUSB0 -b115200 -D -Uflash:w:./..arduino.avr.nano.cpu=atmega328.hex:i
```

Even when overriding the Baud rate in `~/.avrduderc`, this has no effect.

I can successfully flash my Arduino Nano like this:

```
avrdude -v -patmega328p -carduino -P/dev/ttyUSB0 -b57600 -D -Uflash:w:..arduino.avr.nano.cpu=atmega328.hex:i
```

Could you remove the `-b115200` from the commandline when invoking `avrdude`?
The Arduino IDE also doesn't pass that flag:

[image]

I suggest to spare the flag (as the default configuration file for `avrdude` already specifies the correct Baud rate).

per1234, 2018-10-02:
The avrdude configuration file is not going to have the communication speed of the bootloader on your board.

You simply need to use the correct FQBN for your old or clone Nano:
```
arduino:avr:nano:cpu=atmega328old
```
The FQBN you're using is for the new official Nanos, which have a bootloader that communicates at 115200.

Prior99, 2018-10-03:
Thank you for your quick response.
I will try this later and close the issue.

closed 2018-10-09 by per1234