arduino/arduino-cli #62: panic: assignment to entry in nil map

SebCanet, 2018-10-08:
Hi,
I've just tested arduino-cli on a Windows7x64. I follow instruction one by one with no problem, I can compile, but I cannot upload.
To compile I type:
arduino-cli compile --fqbn arduino:avr:uno ../sketchbook/blank
To upload, I type:
arduino-cli upload -p COM13 --fqbn arduino:avr:uno ../sketchbook/blank
and I get back:

panic: assignment to entry in nil map

goroutine 1 [running]:
github.com/arduino/arduino-cli/vendor/github.com/arduino/go-properties-orderedmap.(*Map).Set(0x135cfe6c, 0x137000b2, 0x4, 0x137000b7, 0x12)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/arduino/go-properties-orderedmap/properties.go:226 +0x76
github.com/arduino/arduino-cli/vendor/github.com/arduino/go-properties-orderedmap.(*Map).Merge(0x135cfe6c, 0x135cfe58, 0x1, 0x1, 0x134b8eb8)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/arduino/go-properties-orderedmap/properties.go:396 +0x7d
github.com/arduino/arduino-cli/commands/upload.run(0x13609540, 0x134f7aa0, 0x1, 0x5)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/commands/upload/upload.go:165 +0x3ac
github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra.(*Command).execute(0x13609540, 0x134f7a70, 0x5, 0x6, 0x13609540, 0x134f7a70)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra/command.go:760 +0x20e
github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra.(*Command).ExecuteC(0x13512f00, 0x13608500, 0x13513a40, 0x13513680)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra/command.go:846 +0x28a
github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra.(*Command).Execute(0x13512f00, 0x8f26b8, 0x8f3494)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra/command.go:794 +0x21
main.main()
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/main.go:30 +0x1f

If you have any idea...thanks a lot.

cmaglie, 2018-10-08:
Uhm... this is a nasty bug in the latest 0.3.0 alpha. I'm looking into it, in the meantime you can try the previous version: https://github.com/arduino/arduino-cli/releases

closed 2018-10-08 by cmaglie

cmaglie, 2018-10-08:
0.3.1 with a fix has been released.

SebCanet, 2018-10-08:
Perfect! Works fine!
I've almost finished my 'Electron' Blockly@rduino: [https://github.com/technologiescollege/BlocklyArduino_electrified](https://github.com/technologiescollege/BlocklyArduino_electrified)

thanks a lot!

commit 5fc67871 (2018-10-08): Fixed nasty regression in properties map creation

Fix #62