arduino/arduino-cli #63: `Arduino uno` cannot upload from `raspberry pi 2 model B`

quchunguang, 2018-10-09:
I worked well connecting to my PC with Ubuntu.

And I successfully installed the `auruino-cli` on my  `raspberry pi 2 model B`.
The compile is OK, but when I `arduino-cli upload -p /dev/ttyUSB0 --fqbn arduino:avr:uno MyFirstSketch`,
it shows below uncatched error messages...

```
panic: assignment to entry in nil map

goroutine 1 [running]:
github.com/arduino/arduino-cli/vendor/github.com/arduino/go-properties-orderedmap.(*Map).Set(0x1249be64, 0x1257a0b2, 0x4, 0x1257a0b7, 0x12)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/arduino/go-properties-orderedmap/properties.go:226 +0x6c
github.com/arduino/arduino-cli/vendor/github.com/arduino/go-properties-orderedmap.(*Map).Merge(0x1249be64, 0x1249be50, 0x1, 0x1, 0x123018c8)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/arduino/go-properties-orderedmap/properties.go:396 +0x7c
github.com/arduino/arduino-cli/commands/upload.run(0x124c9900, 0x12397b00, 0x1, 0x5)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/commands/upload/upload.go:165 +0x364
github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra.(*Command).execute(0x124c9900, 0x12397ad0, 0x5, 0x6, 0x124c9900, 0x12397ad0)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra/command.go:760 +0x20c
github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra.(*Command).ExecuteC(0x123c72c0, 0x124c88c0, 0x123c7e00, 0x123c7a40)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra/command.go:846 +0x20c
github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra.(*Command).Execute(0x123c72c0, 0x3effc, 0x5140c0)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra/command.go:794 +0x1c
main.main()
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/main.go:30 +0x18
```
BTW, it is worked fine with the `arduino+arduino.mk` CLI solutions on this raspberry for years.

per1234, 2018-10-09:
Which version of arduino-cli are you using? There is a similar issue here: https://github.com/arduino/arduino-cli/issues/62, which was already fixed in today's 0.3.1-alpha.preview release:
https://github.com/arduino/arduino-cli/releases/tag/0.3.1-alpha.preview

quchunguang, 2018-10-09:
> Which version of arduino-cli are you using? There is a similar issue here: #62, which was already fixed in today's 0.3.1-alpha.preview release:
> https://github.com/arduino/arduino-cli/releases/tag/0.3.1-alpha.preview

Linux ARM 0.3.1-alpha.preview
download from home page here(not by `go get`)

quchunguang, 2018-10-09:
It worked after i just `go get` to current 98fec8b

closed 2018-10-09 by quchunguang