arduino/arduino-cli #64: Can't upload compiled code on esp8266

brusherru, 2018-10-10:
I want to upload code on esp8266 (wifi_slot).
I already applied patch from [#35](https://github.com/arduino/arduino-cli/issues/35) and it works, so I can compile code, but I can't upload it... 

## Env
- Windows 10
- arduino-cli@0.2.2

This bug can't be reproduced on Linux/MacOS.

## Steps to reproduce
1. Create and save sketch in the sketchbook directory.
2. Compile it for the esp8266:esp8266:wifi_slot using properties in the FQBN (Why not `--build-properties` see below):
  `arduino-cli --fqbn esp8266:esp8266:wifi_slot:CpuFrequency=80,Vtable=flash,FlashFreq=40,FlashMode=qio,FlashSize=1M0,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None,FlashErase=none,UploadSpeed=921600 compile C:\Users\the-awesome-bob\arduino-projects\my-sketchbook\my_sketch`
3. Upload it:
    `arduino-cli --fqbn esp8266:esp8266:wifi_slot:CpuFrequency=80,Vtable=flash,FlashFreq=40,FlashMode=qio,FlashSize=1M0,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None,FlashErase=none,UploadSpeed=921600 --port COM5 upload C:\Users\the-awesome-bob\arduino-projects\my-sketchbook\my_sketch`

## Expected behavior
It uploaded and works fine.

## Actual behavior
Arduino-cli exits with error:
```
No new serial port detected.
error: stat C:\Users\the-awesome-bob\arduino-projects\my-sketchbook\my_sketch/my_sketch.esp8266.esp8266.wifi_slot.CpuFrequency=80,Vtable=flash,FlashFreq=40,FlashMode=qio,FlashSize=1M0,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None,FlashErase=none,UploadSpeed=921600.bin failed: No such file or directory
Error: exit status 2
Error during upload.
```

When I replace the path with UNC path `\\localhost\c$\Users\...` it exists with:
```
Compiled sketch not found. Please compile first.
```

## Why not `--build-properties`
`upload` command does not have a special flag to specify upload properties. So it can't upload sketch even if you successfully compiled it with `--build-properties`.

## Why it happens?
1. First of all, you can see a wrong path separator in the path: "...sketchbook\my_sketch**/**my_sketch.esp8266..."
2. The full path is longer than 260 symbols, so maybe this is the problem on Windows.

## How to fix it
I propose two ways to fix it:
1. Just remove properties from the filename (`my_sketch.esp8266.esp8266.wifi_slot.bin`).
2. Add a special flag to `upload` command.

embhobbb, 2018-10-14:
i am also facing same problem.

brusherru, 2018-10-19:
@embhobbb place thumb up in this case.
Probably, it will help the developers to measure how important this task is ;)

embhobbb, 2018-10-21:
@brusherru  Did you get any solution yet?

cmaglie, 2018-10-26:
Ok it seems that the problem is the path of the compiled `sketch.bin` that is too long. Some improvements that comes to mind:

1) remove the "options" part from the filename and call the file just `my_sketch.esp8266.esp8266.wifi_slot.bin` instead of `my_sketch.esp8266.esp8266.wifi_slot.CpuFrequency=80,Vtable=flash,FlashFreq=40,FlashMode=qio,FlashSize=1M0,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None,FlashErase=none,UploadSpeed=921600.bin`.

2) make the `upload` command become "compile+upload". This would mimic the IDE behavior and it will make the cli tolerant to cores that do not define a property to export the compiled binary.

3) add options to specify file name like `arduino-cli compile --output myname.bin` and `arduino-cli upload --input myname.bin`

I'd implement 1) to begin with, but also 2) and 3) may be part of the next alpha as well.

brusherru, 2018-10-30:
@cmaglie it works! 👏

PR #79 (2018-10-30): Various improvements in compile / upload
- The exported compiled binary now does not contains the FQBN "config" part anymore. This means that, for example, the default exported file for `arduino:avr:mega:cpu=atmega1280` will be called `Blink.arduino.avr.mega.hex` instead of `Blink.arduino.avr.mega.cpu=atmega1280.hex`.

The change above should fix #64

- The `compile` command now has a `-o, --output` to manually specifiy the output filename for the compiled binary.
- The `upload` command now has a `-i, --input` to manually specify the input filename of the binary to upload to the board.
commit 76d471d9 (2018-10-30): Do not use FQBN "configs" part in saving .hex or .bin from compile

Fix #64
closed 2018-10-30 by cmaglie