arduino/arduino-cli #65: Regression in 0.3.1 - can't upload on any board

brusherru, 2018-10-10:
Installing, compiling and etc works fine.

But when I try to upload compiled sketch onto my board (`arduino:avr:leonardo`) it throws:
```
panic: assignment to entry in nil map

goroutine 1 [running]:
github.com/arduino/arduino-cli/vendor/github.com/arduino/go-properties-orderedmap.(*Map).Set(0x11f1be6c, 0x120120b2, 0x4, 0x120120b7, 0x12)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/arduino/go-properties-orderedmap/properties.go:226 +0x76
github.com/arduino/arduino-cli/vendor/github.com/arduino/go-properties-orderedmap.(*Map).Merge(0x11f1be6c, 0x11f1be58, 0x1, 0x1, 0x11e04e50)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/arduino/go-properties-orderedmap/properties.go:396 +0x7d
github.com/arduino/arduino-cli/commands/upload.run(0x11f53680, 0x11e43aa0, 0x1, 0x5)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/commands/upload/upload.go:165 +0x3ac
github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra.(*Command).execute(0x11f53680, 0x11e43a70, 0x5, 0x6, 0x11f53680, 0x11e43a70)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra/command.go:760 +0x20e
github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra.(*Command).ExecuteC(0x11e5f040, 0x11f52640, 0x11e5fb80, 0x11e5f7c0)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra/command.go:846 +0x28a
github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra.(*Command).Execute(0x11e5f040, 0x8f26b8, 0x8f3494)
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/vendor/github.com/spf13/cobra/command.go:794 +0x21
main.main()
        /home/jenkins/workspace/arduino-cli/src/github.com/arduino/arduino-cli/main.go:30 +0x1f
```

cmaglie, 2018-10-11:
I should have fixed this in 0.3.1 :fearful:, let me do a quick check...

brusherru, 2018-10-11:
@cmaglie it looks like 0.3.1 differs from 0.3.0 only by changes in the README.md 🤔

per1234, 2018-10-11:
@brusherru here's the important commit between 0.3.0-alpha.preview and 0.3.1-alpha.preview that should have fixed this problem (it did for me at least): https://github.com/arduino/arduino-cli/commit/5fc6787168de846e76f7bf71b1c452abb3bb4614

cmaglie, 2018-10-11:
@brusherru I've just tried 0.3.1 on a windows machine and it seems to work as expected, may you double check if it is really a 0.3.1 and not a 0.3.0? try to run `arduino-cli version` to be 100% sure.

brusherru, 2018-10-11:
@cmaglie @per1234 oh, you're right. It's my failure. Sorry 🙈

closed 2018-10-11 by brusherru