arduino/arduino-cli #66: Wrong core.a location for WAVGAT platform

grafalex82, 2018-10-10:
Hello and thank you for the great tool!

Recently I bought an arduino-nano clone on an lgt8F328d chip which is chinease ATMega328 clone with some improvements. Since this is not a 1-to-1 clone they provide a version of arduino stuff. 

https://drive.google.com/open?id=10gwrG9uTDwaEO-7EudsmBkfgdcyrcABI

I tested this on Windows with normal arduino IDE and it seems working fine. The package consists of text files, so it is safe to download. Now I am trying to get it working on Linux (actually Raspbian)

First I had hard time installing this to arduino-cli (at least uncertainty where to put those files). I ended up putting update/hardware/WAV/avr contents from the archive to .arduino15/packages/arduino/hardware/WAV/1.0.0 as arduino-cli wants a version number in the path. Eventually it picked up list of boards

But the next problem looks more serious (at least after 2 days fighting with it I still have no solution).

When I try to compile a regular blink sketch for traditional arduino:avr:nano board build process creates core.a at /tmp/arduino-sketch-098F6BCD4621D373CADE4E832627B4F6/**core/**core.a

When I try to compile for arduino:WAV:lardu_328e arduino-cli calls avr-ar with output path set to /tmp/arduino-sketch-098F6BCD4621D373CADE4E832627B4F6/core.a (no 'core' subdirectory). Later link can't find core.a and build fails. 

I tried to find the answer in platforms.txt as well as in arduino-cli sources, but no luck. It looks odd as ar and linker commands both use the same archive {build.path}/{archive_file} path:

```
## Create archives
#recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}"

## Combine gc-sections, archives, and objects
recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags} -mmcu={build.mcu} {compiler.c.elf.extra_flags} -o "{build.path}/{build.project_name}.elf" {object_files} "{build.path}/{archive_file}" "-L{build.path}" -lm
```

I dumped exact arguments passed to ar and linker:

```
/home/pi/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-ar
rcs
/tmp/arduino-sketch-098F6BCD4621D373CADE4E832627B4F6/core.a

...

/home/pi/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-gcc
-w
-Os
-Wl,--gc-sections
-mmcu=atmega328p
-o
/tmp/arduino-sketch-098F6BCD4621D373CADE4E832627B4F6/test.ino.elf
/tmp/arduino-sketch-098F6BCD4621D373CADE4E832627B4F6/sketch/test.ino.cpp.o
/tmp/arduino-sketch-098F6BCD4621D373CADE4E832627B4F6/core/core.a
-L/tmp/arduino-sketch-098F6BCD4621D373CADE4E832627B4F6
-lm
```

Can someone help me?

brusherru, 2018-10-11:
It looks like a problem with the esp8266 `core.a` ([solution](https://github.com/arduino/arduino-cli/issues/47#issuecomment-424666516)).
Try replace `{build.path}/{archive_file}` with `{archive_file_path}`.

grafalex82, 2018-10-11:
Thanks Kirill,

That fixed my issue. I also had to copy one more change from arduino's platform.txt:

```
## Save hex
recipe.output.tmp_file={build.project_name}.hex
recipe.output.save_file={build.project_name}.{build.variant}.hex
```

closed 2018-10-20 by grafalex82