arduino/arduino-cli #74: make importable Go package

gordonklaus, 2018-10-22:
I would like to use the functionality of this project from a Go program without having to exec a new process, pass text input, and parse text output.  The main benefit would be to have a statically typed API to work against.  It would also make it easier to depend on a specific version of the package (whose API may change over time).

Ideally such a package would not be called `arduino-cli` but perhaps `arduino-go`.  The `arduino-cli` package would then be a thin wrapper around that package, responsible only for marshaling commands and data.

cmaglie, 2018-10-25:
That was the purpose of the `arduino` package, I've tried to put as much as possible of the business logic there, but I see that it's still too low-level and calling the cli "commands" directly may be convenient.

We need a similar interface too, so I don't exclude that it will be implemented at some point, but for now the priority is to fix bugs and stabilize the API.

masci, 2019-08-05:
A major refactoring invovled extracting a `commands` package that's used by both the command line and the gRPC interface. The abstraction it provides should be good enough to execute several operations programmatically, feel free to open a more specific issue if the API provided isn't flexible enough for your needs.

closed 2019-08-05 by masci