arduino/arduino-cli #77: Feature request: Dump env vars: arduino-cli setenv --fqbn my:precious:board

marcond, 2018-10-25:
I'm just testing `arduino-cli` and I think it's GREAT. It will simplify so many things.

When I went to test a bootloader I'm working on, I missed an option to output a proper set of environment vars for compilation, matching a given architecture. Something generating an output like this:
```
$ arduino-cli setenv --fqbn arduino:avr:uno
           -- or --
$ arduino-cli config dump --fqbn arduino:avr:uno
# Vars for arduino:avr:uno
# Generated by arduino-cli-0.3.1-alpha.preview
AVR_GCC_HOME=~/.arduino15/packages/arduino/tools/avr-gcc/5.4.0-atmel3.6.1-arduino2
export AVR_GCC_HOME
PATH=$AVR_GCC_HOME/bin:$PATH
export PATH
MCU=atmega328p
export MCU
F_CPU=16000000L
export F_CPU
INCLUDES=~/.arduino15/packages/arduino/hardware/avr/1.6.23/cores/arduino:~/.arduino15/packages/arduino/hardware/avr/1.6.23/variants/standard
export INCLUDES
# ....and so on....
```
So I could just `arduino-cli setenv --fqbn arduino:avr:uno >myvars.sh` and then `. myvars.sh` afterwards. VERY handy. 

I could hack something around this, but I'm lacking free time for some weeks.

Thanks.

bxparks, 2018-10-25:
(Drive-by comment...)

I am not sure using environment variables is the best way to achieve this, because it would be too difficult to maintain and debug so many environment variables that might affect the behavior of the command line tools. However, a better place to capture this information might be a config file, say `.cli-config.yml`.

But you would probably want to capture multiple sets of such parameters. Then it looks like you are asking for something that is similar to the concept of "environments" in the `platformio.ini` files of PlatformIO.

marcond, 2018-11-05:
Right now I'm looking into the `board details` feature (#45) just added. It seems that I can use that option to retrieve all the data I need, let's see!

The .cli-config.yml was tempting, however I felt that it might change over time and break things. So I would not like to sneak into cli-config to fetch parameters. Right now I'm testing some options on my own project. About the number of env vars, actually I was not thinking about "all" of them, just the important ones (line tools path, MCU, F_CPU and a few others).

I'm not familiar with environments from platformio, but I'll take a look.

masci, 2019-08-16:
Closing as `board details` seems to have fixed the issue.

closed 2019-08-16 by masci