arduino/arduino-cli #78: Feature request: update libraries

SebCanet, 2018-10-26:
Hi,
as we can make a 'update-index', is it possible to 'update-lib' ?
Thanks, this tool is very useful!

bxparks, 2018-10-26:
Isn't `arduino-cli lib upgrade` what you are asking for? (The word `upgrade` is a bit unfortunate, since the Linux `apt` tool makes a distinction between `upgrade` and `update`, it confused me for a split second.)

SebCanet, 2018-10-26:
Thansk! It was not in documentation, isn't it?

closed 2018-10-26 by SebCanet

bxparks, 2018-10-26:
Oh, I meant to type something different 2 posts up: The `arduino-cli` command actually follows the convention of the `apt` tool by using `upgrade` instead of `update`. It's the Library Manager in the IDE that shows an "Update" button to mean 'upgrade' in the arduino-cli tool. (I knew there was a small conflict in words somewhere, my fingers typed faster than my brain could remember.)