arduino/arduino-cli #83: Upload tool 'avrdude' not found.

RubberChicken69, 2018-10-31:
Have tried fresh install a number of times on Debian 9.
This is a testing setup under DietPi in Virtualbox
Always get the same result:

Upload tool 'avrdude' not found.
root@Testing:/mnt/dietpi_userdata# ./arduino-cli upload -p /dev/ttyUSB0 --verbose --debug --fqbn Moteino:avr:MoteinoMEGA /mnt/dietpi_userdata/Arduino/test
INFO[0000] Initiating configuration
INFO[0000] Unserializing configurations from /mnt/dietpi_userdata/.cli-config.yml
INFO[0000] Checking if CLI is Bundled into the IDE
INFO[0000] Candidate IDE Directory: /mnt/dietpi_userdata
INFO[0000] CLI is not bundled into the IDE
INFO[0000] Configuration set
INFO[0000] Starting root command preparation (`arduino`)
INFO[0000] Formatter set
INFO[0000] Initializing package manager
INFO[0000] Loading hardware from: /mnt/dietpi_userdata/.arduino15/packages
INFO[0000] Loading package Moteino from: /mnt/dietpi_userdata/.arduino15/packages/Moteino/hardware
INFO[0000] Loaded platform                               platform="Moteino:avr@1.4.0"
INFO[0000] Loading package arduino from: /mnt/dietpi_userdata/.arduino15/packages/arduino/hardware
INFO[0000] Loaded platform                               platform="arduino:avr@1.6.23"
INFO[0000] Checking existence of 'tools' path: /mnt/dietpi_userdata/.arduino15/packages/arduino/tools
INFO[0000] Loading tools from dir: /mnt/dietpi_userdata/.arduino15/packages/arduino/tools
INFO[0000] Loaded tool                                   tool="arduino:arduinoOTA@1.2.1"
INFO[0000] Loaded tool                                   tool="arduino:avr-gcc@4.8.1-arduino5"
INFO[0000] Loaded tool                                   tool="arduino:avr-gcc@5.4.0-atmel3.6.1-arduino2"
INFO[0000] Loaded tool                                   tool="arduino:avrdude@6.0.1-arduino5"
INFO[0000] Loaded tool                                   tool="arduino:avrdude@6.3.0-arduino14"
INFO[0000] Loading package builtin from: /mnt/dietpi_userdata/.arduino15/packages/builtin
INFO[0000] Checking existence of 'tools' path: /mnt/dietpi_userdata/.arduino15/packages/builtin/tools
INFO[0000] Loading tools from dir: /mnt/dietpi_userdata/.arduino15/packages/builtin/tools
INFO[0000] Loaded tool                                   tool="builtin:ctags@5.8-arduino11"
ERRO[0000] Upload tool 'avrdude' not found.              error="<nil>"
Upload tool 'avrdude' not found.

cmaglie, 2018-10-31:
May you post your configuration? in particular the 3rd party hardware URL you are using.

RubberChicken69, 2018-10-31:
This issue does not appear to be board specific, I tried with some other boards that I have lying around and while the upload tools all seem to be present, arduino-cli will not upload:

root@Testing:~# arduino-cli board list
FQBN                    Port            ID              Board Name
arduino:avr:mega        /dev/ttyACM0    2341:0010       Arduino/Genuino Mega or Mega 2560

root@Testing:~# arduino-cli upload --debug -p /dev/ttyACM0 --fqbn arduino:avr:mega /mnt/dietpi_userdata/Arduino/test3
INFO[0000] Initiating configuration
INFO[0000] Unserializing configurations from /usr/local/bin/.cli-config.yml
INFO[0000] Checking if CLI is Bundled into the IDE
INFO[0000] Candidate IDE Directory: /usr/local/bin
INFO[0000] CLI is not bundled into the IDE
INFO[0000] Configuration set
INFO[0000] Starting root command preparation (`arduino`)
INFO[0000] Formatter set
INFO[0000] Initializing package manager
INFO[0000] Loading hardware from: /mnt/dietpi_userdata/.arduino15/packages
INFO[0000] Loading package Moteino from: /mnt/dietpi_userdata/.arduino15/packages/Moteino/hardware
INFO[0000] Loaded platform                               platform="Moteino:avr@1.4.0"
INFO[0000] Loading package arduino from: /mnt/dietpi_userdata/.arduino15/packages/arduino/hardware
INFO[0000] Loaded platform                               platform="arduino:avr@1.6.23"
INFO[0000] Checking existence of 'tools' path: /mnt/dietpi_userdata/.arduino15/packages/arduino/tools
INFO[0000] Loading tools from dir: /mnt/dietpi_userdata/.arduino15/packages/arduino/tools
INFO[0000] Loaded tool                                   tool="arduino:arduinoOTA@1.2.1"
INFO[0000] Loaded tool                                   tool="arduino:arm-none-eabi-gcc@4.8.3-2014q1"
INFO[0000] Loaded tool                                   tool="arduino:avr-gcc@4.8.1-arduino5"
INFO[0000] Loaded tool                                   tool="arduino:avr-gcc@5.4.0-atmel3.6.1-arduino2"
INFO[0000] Loaded tool                                   tool="arduino:avrdude@6.0.1-arduino5"
INFO[0000] Loaded tool                                   tool="arduino:avrdude@6.3.0-arduino14"
INFO[0000] Loading package builtin from: /mnt/dietpi_userdata/.arduino15/packages/builtin
INFO[0000] Checking existence of 'tools' path: /mnt/dietpi_userdata/.arduino15/packages/builtin/tools
INFO[0000] Loading tools from dir: /mnt/dietpi_userdata/.arduino15/packages/builtin/tools
INFO[0000] Loaded tool                                   tool="builtin:ctags@5.8-arduino11"
INFO[0000] Loading package stm32duino from: /mnt/dietpi_userdata/.arduino15/packages/stm32duino/hardware
INFO[0000] Loaded platform                               platform="stm32duino:STM32F1@2018.9.24"
INFO[0000] Checking existence of 'tools' path: /mnt/dietpi_userdata/.arduino15/packages/stm32duino/tools
INFO[0000] Loading tools from dir: /mnt/dietpi_userdata/.arduino15/packages/stm32duino/tools
INFO[0000] Loaded tool                                   tool="stm32duino:stm32tools@2018.9.24"
ERRO[0000] Error during upload.                          error="exit status 1"
Error: exit status 1
Error during upload.
root@Testing:~#

root@Testing:/mnt/dietpi_userdata/Arduino/test2# arduino-cli compile --fqbn stm32duino:STM32F1:genericSTM32F103C /mnt/dietpi_userdata/Arduino/test2
Sketch uses 14356 bytes (21%) of program storage space. Maximum is 65536 bytes.
Global variables use 3088 bytes (15%) of dynamic memory, leaving 17392 bytes for local variables. Maximum is 20480 bytes.
root@Testing:/mnt/dietpi_userdata/Arduino/test2# arduino-cli upload -p /dev/ttyACM0 --fqbn stm32duino:STM32F1:genericSTM32F103C /mnt/dietpi_userdata/Arduino/test2
Upload tool 'maple_upload' not found.
root@Testing:/mnt/dietpi_userdata/Arduino/test2# arduino-cli upload --debug -p /dev/ttyACM0 --fqbn stm32duino:STM32F1:genericSTM32F103C /mnt/dietpi_userdata/Arduino/test2
INFO[0000] Initiating configuration
INFO[0000] Unserializing configurations from /usr/local/bin/.cli-config.yml
INFO[0000] Checking if CLI is Bundled into the IDE
INFO[0000] Candidate IDE Directory: /usr/local/bin
INFO[0000] CLI is not bundled into the IDE
INFO[0000] Configuration set
INFO[0000] Starting root command preparation (`arduino`)
INFO[0000] Formatter set
INFO[0000] Initializing package manager
INFO[0000] Loading hardware from: /mnt/dietpi_userdata/.arduino15/packages
INFO[0000] Loading package Moteino from: /mnt/dietpi_userdata/.arduino15/packages/Moteino/hardware
INFO[0000] Loaded platform                               platform="Moteino:avr@1.4.0"
INFO[0000] Loading package arduino from: /mnt/dietpi_userdata/.arduino15/packages/arduino/hardware
INFO[0000] Loaded platform                               platform="arduino:avr@1.6.23"
INFO[0000] Checking existence of 'tools' path: /mnt/dietpi_userdata/.arduino15/packages/arduino/tools
INFO[0000] Loading tools from dir: /mnt/dietpi_userdata/.arduino15/packages/arduino/tools
INFO[0000] Loaded tool                                   tool="arduino:arduinoOTA@1.2.1"
INFO[0000] Loaded tool                                   tool="arduino:arm-none-eabi-gcc@4.8.3-2014q1"
INFO[0000] Loaded tool                                   tool="arduino:avr-gcc@4.8.1-arduino5"
INFO[0000] Loaded tool                                   tool="arduino:avr-gcc@5.4.0-atmel3.6.1-arduino2"
INFO[0000] Loaded tool                                   tool="arduino:avrdude@6.0.1-arduino5"
INFO[0000] Loaded tool                                   tool="arduino:avrdude@6.3.0-arduino14"
[... cut ...]
INFO[0000] Loaded platform                               platform="stm32duino:STM32F1@2018.9.24"
INFO[0000] Checking existence of 'tools' path: /mnt/dietpi_userdata/.arduino15/packages/stm32duino/tools
INFO[0000] Loading tools from dir: /mnt/dietpi_userdata/.arduino15/packages/stm32duino/tools
INFO[0000] Loaded tool                                   tool="stm32duino:stm32tools@2018.9.24"
ERRO[0000] Upload tool 'maple_upload' not found.         error="<nil>"
Upload tool 'maple_upload' not found.

PR #103 (2019-01-07): Various fix and improvements to "upload"
- Added tests for upload command (parsing only)
- Added check for missing "recipe.output.tmp_file" property
- Fixed check for `upload.tool` existence (it should not check for tool existence but just for the referenced platform, if specified). Should help to fix #83 

Not strictly related to upload, but useful to make tests run more efficiently:
- Downloaded archives are kept in a temporary folder and reused (so they are downloaded only once and not many times)
closed 2019-01-07 by cmaglie