arduino/arduino-cli #86: Can't use standard STL includes in library code

grafalex82, 2018-11-03:
When trying to compile DMStepper library arduino-cli hit #include <vector> (see [https://github.com/toxnico/DMStepper/blob/master/src/dmsteppergroup.h](https://github.com/toxnico/DMStepper/blob/master/src/dmsteppergroup.h)) and tried to interpret this as an another library dependency

```
Alternatives for vector: []
ResolveLibrary(vector)
  -> candidates: []
In file included from /home/pi/Arduino/libraries/DMStepper/src/dmsteppergroup.cpp:1:0:
/home/pi/Arduino/libraries/DMStepper/src/dmsteppergroup.h:4:18: fatal error: vector: No such file or directory
 #include <vector>
                  ^
compilation terminated.

```

GeraldR63, 2018-12-27:
Hello Grafalex,

to be able to compile dmstepper it's required to add line below to your sketch:

// ArduiniSTL required in front of dmstepper.h.  Probably you have to install this 
#include <ArduinoSTL.h>
#include <dmstepper.h>

Than you can compile and upload.

regards
Gerald

grafalex82, 2018-12-28:
Thanks. I'll try this

facchinm, 2019-01-02:
Closing as fixed

closed 2019-01-02 by facchinm