arduino/arduino-cli #88: offfical Arduino libraries not seen

SebCanet, 2018-11-06:
Hi,
with 0.31.preview on Windows, when I type arduino-cli lib list, it lists only libraries inside my sketchbook/libraries folder.
I changed ".cli-config.yml" like this:

libraries:
- ../sketchbook/libraries
- ./libraries

But when I compile it doesn't find Servo.h

Thanks!

masci, 2019-08-05:
Can you confirm it's still the case with latest 0.4.0?

masci, 2019-08-16:
Closing as stale, feel free to reopen should you have the same issue again.

SebCanet, 2019-08-27:
Still the same pb. Even if it's in a subfolder:
```
libraries:
    - ../libraries
    - ../sketchbook/libraries
```
My folders are like that:
project
|-arduino (with arduino-cli.exe inside)
|-libraries (with liraries from Arduino IDE)
|-sketchbook
   |-libraries (fro personnal lib)
   |-sketches

SebCanet, 2019-09-05:
Still the same pb with v0.5.0
```
libraries:
    - ./libraries
    - ../sketchbook/libraries
```

masci, 2019-09-05:
Thanks for confirming @SebCanet

ubidefeo, 2020-05-27:
@SebCanet 
I endorse this.
I believe the config file(s) should allow for a deeper degree of behaviour customisation.
@cmaglie let's look at it in our next session, I have a few ideas :)

SebCanet, 2020-05-27:
Great 👍