arduino/arduino-cli #92: Provide a way to query only usb boards

hgrecco, 2018-11-21:
In many cases I know that my arduino board is connected via USB (not network). It seems that is currently not possible to look only for usb boards using `arduino-cli board list`. This would be convenient to reduce the discovery time.

flatsiedatsie, 2018-12-24:
This would also make me very happy. I am currently developing a small web-based tool around the Arduino-CLI, and this option would make things more predictable. In the interface users are prompted to plug in their arduino. It would be odd if network based devices could 'magically' show up without them having plugged anything into the USB port yet.

umbynos, 2022-10-12:
In the meantime we added back the network discovery, the defualt discovery timout is 1 second, but if you like you can specify it with `--discovery-timeout` flag. For example `arduino-cli board list --discovery-timeout 0.1s` this is really fast.

closed 2022-10-12 by umbynos