arduino/arduino-cli #96: JSON Response to lib Commands

GItThatSparrow, 2018-12-02:
**AS A** User

**I NEED TO** determine the result of a call to the `arduino-cli` based on the response
   **AND** the command specifies the `--format=json` parameter

**SO THAT** I can distinguish the difference in between a successful operation and a failed operation

--------

# Example
I issue a command like
```
>> arduino-cli lib install AudioZero@1.0.0 --format=json
```

But the response is empty, like:
```
>> arduino-cli lib install AudioZero@1.0.0 --format=json

>>
```

## Equivalent example
Same result when I issue a command like
```
>> arduino-cli lib uninstall AudioZero@1.0.0 --format=json
```

But the response is empty, like:
```
>> arduino-cli lib uninstall AudioZero@1.0.0 --format=json

>>
```

mastrolinux, 2019-07-17:
@philippejadin you can probably now use the new GRPC interface. This bug is still confirmed and I am not use if best to return an empty json or something like `{'success': 'true', 'message': 'some message here or empty string'}`

masci, 2019-07-18:
I think when invoking the CLI process the source of truth should always be the process exit code.

If the exit code is `0` operation was successful, otherwise you should make no assumptions about what's in the standard error/output: instead, you would look at the meaning of the exit code. Think about the CLI exiting because of a `panic` when likely the program wasn't able to properly print any output message, let alone a JSON structured message.

Different topic is whether printing an empty JSON object `{}` or nothing at all when the command succeeds but the result is empty. I personally like the `docker` approach, where nothing is printed if the response was empty (try doing `docker ps  --format='{{json .}}'` vs `docker ps` when no containers are running).

mastrolinux, 2019-07-19:
ok to have completely no output, it respects the UNIX phylosophy. I believe now the main issue is some commands returns code different from 0 even if the command run successfully (i.e. running twice lib install with the same libs)

masci, 2019-08-28:
The expected behavior is:

* exit code of the CLI determines the status of the operation requested (success/fail)
* JSON output is only considered valid when the exit code is 0

I'm going to close this as what we have in `master` works as described.

closed 2019-08-28 by masci