arduino/arduino-cli #105: Version restriction for installing library

scls19fr, 2019-01-04:
Hello,

I noticed in doc:

     arduino-cli lib install "WiFi101"

install latest "published" library named `WiFi101`.

I wonder if it's currently possible to restrict version of library to install a library using a command line like

     arduino-cli lib install "ArduinoJson<6.0.0&&>=5.0.0"

or simply

     arduino-cli lib install "ArduinoJson<6.0.0,>=5.0.0"

Such a command could install latest version between 5.0.0 (include) and 6.0.0 (exclude)
ie, currently 5.13.4.

This is quite similar to Python requirements.txt syntax.
See https://www.python.org/dev/peps/pep-0440/#compatible-release and https://stackoverflow.com/questions/8795617/how-to-pip-install-a-package-with-min-and-max-version-range

Installing a given version should also be considered (or documented)

     arduino-cli lib install "ArduinoJson==5.13.4"

Specifying an "unpublished" library could also be supported (with care)

https://stackoverflow.com/questions/16584552/how-to-state-in-requirements-txt-a-direct-github-source

I also wonder if the concept of "environment" (comparable to [conda environments](https://conda.io/docs/user-guide/tasks/manage-environments.html) or [Julia Pkg environments](https://docs.julialang.org/en/v1/stdlib/Pkg/index.html)) have been considered (it could help to have several version of library installed without odd effects).

Kind regards

PS : related issue https://github.com/arduino/Arduino/issues/5795

scls19fr, 2019-01-04:
s/published/registered/g
s/unpublished/unregistered/g

phd, 2021-02-22:
> Installing a given version should also be considered (or documented)
> 
> ```
>  $ arduino-cli lib install "ArduinoJson==5.13.4"
> ```

You can install a specific version of a library with:

```
$ arduino-cli lib install "ArduinoJson@5.13.4"
```

scls19fr, 2021-02-22:
Thanks @phd . That's an interesting improvement of arduino-cli.
But I don't think version restriction are currently implemented (ie lib version number lower than a given version number and greater than an other one)

ubidefeo, 2021-02-22:
@silvanocerza do we support this?
Might be an interesting one if we don't :)

silvanocerza, 2021-02-23:
I like this, would be something nice to have. I think it would be feasible but I'd have to investigate a bit first just to be sure.