arduino/arduino-cli #107: arduino-cli lib search should display nearest names

scls19fr, 2019-01-05:
Hello,

Currently arduino-cli can find a library when searched name match exactly lib name (lowercased):

```
$ ./arduino-cli lib search ArduinoJSON
Name: "ArduinoJson"
  Author:  Benoit Blanchon <blog.benoitblanchon.fr>
  Maintainer:  Benoit Blanchon <blog.benoitblanchon.fr>
  Sentence:  An efficient and elegant JSON library for Arduino.
  Paragraph:  ArduinoJson supports ✔ serialization, ✔ deserialization, ✔ MessagePack, ✔ fixed allocation, ✔ zero-copy, ✔ streams, and more. It is the most popular Arduino library on GitHub ❤❤❤❤❤. Check out arduinojson.org for a comprehensive documentation.
  Website:  https://arduinojson.org/?utm_source=meta&utm_medium=library.properties
  Category:  Data Processing
  Architecture:  *
  Types:  Contributed
  Versions:  [4.0.0, 5.0.0, 5.0.1, 5.0.2, 5.0.3, 5.0.4, 5.0.5, 5.0.6, 5.0.7, 5.0.8, 5.1.0-beta.1, 5.1.0-beta.2, 5.1.0, 5.1.1, 5.2.0, 5.3.0, 5.4.0, 5.5.0, 5.5.1, 5.6.0, 5.6.1, 5.6.2, 5.6.3, 5.6.4, 5.6.5, 5.6.6, 5.6.7, 5.7.0, 5.7.1, 5.7.2, 5.7.3, 5.8.0, 5.8.1, 5.8.2, 5.8.3, 5.8.4, 5.9.0, 5.10.0, 5.10.1, 5.11.0, 5.11.1, 5.11.2, 5.12.0, 5.13.0, 5.13.1, 5.13.2, 5.13.3, 5.13.4, 6.0.0-beta, 6.0.1-beta, 6.1.0-beta, 6.2.0-beta, 6.2.1-beta, 6.2.2-beta, 6.2.3-beta, 6.3.0-beta, 6.4.0-beta, 6.5.0-beta, 6.6.0-beta, 6.7.0-beta]
```

But not when it's slightly different

```
$ ./arduino-cli lib search ArduinoJason
No library found matching `arduinojason` search query
```

Maybe Arduino CLI could find nearest library named (or 5 nearest library name)

```
$ ./arduino-cli lib search ArduinoJason
No library found matching `arduinojason` search query
Did you mean?
ArduinoJson
...
```

I did something quite similar in https://github.com/scls19fr/arduino_libraries_search/blob/master/utils.py#L110-L127

using a similarity measurement named [Sørensen–Dice coefficient](https://en.wikipedia.org/wiki/S%C3%B8rensen%E2%80%93Dice_coefficient)

Other metrics / edit distance may be considered
https://en.wikipedia.org/wiki/String_metric

Kind regards

PS : quite similar to this Julia Pkg issue https://github.com/JuliaLang/Pkg.jl/issues/616

scls19fr, 2019-01-05:
https://github.com/danieldk/editdistance may help because it implements [Levenshtein algorithm](https://en.wikipedia.org/wiki/Levenshtein_distance) in Go

flatsiedatsie, 2019-01-10:
Duplicate: https://github.com/arduino/arduino-cli/issues/104

flatsiedatsie, 2019-05-24:
It seems this has been fixed.

per1234, 2019-05-24:
@flatsiedatsie which version of arduino-cli are you using? I still get no search results from `arduino-cli lib search ArduinoJason` using either the tip of the `master` branch on arduino/arduino-cli or the tip of the `daemon` branch on cmaglie/arduino-cli.

flatsiedatsie, 2019-05-30:
@per1234 You're right, I get the same result on that query. But I don't get any results for `ArduinoJason` in the Arduino IDE either? I'm using the latest compiled alpha that I downloaded from Github.

per1234, 2019-05-30:
>But I don't get any results for ArduinoJason in the Arduino IDE either?

That's correct. Unlike your https://github.com/arduino/arduino-cli/issues/104, which is only asking that the search behavior made to work the same as in the Arduino IDE's Library Manager, this issue is a request for an enhancement of the search behavior beyond what is provided by the Arduino IDE.

masci, 2019-12-19:
This library should ease the implementation of this feature a lot https://github.com/imjasonmiller/godice

closed 2020-04-02 by rsora

scls19fr, 2020-04-02:
Thanks @rsora

scls19fr, 2020-04-02:
Thanks @BigHeadGeorge

giripriyadarshan, 2021-04-06:
Also, please make the search whitespace friendly probably.

Because, if I search with proper whitespace, then the search result is perfect

```zsh
> arduino-cli lib search painless mesh
Updating index: library_index.json downloaded                                   
Name: "Painless Mesh"
  Author: Coopdis,Scotty Franzyshen,Edwin van Leeuwen,Germán Martín,Maximilian Schwarz,Doanh Doanh
  Maintainer: Edwin van Leeuwen
  Sentence: A painless way to setup a mesh with ESP8266 and ESP32 devices
  Paragraph: A painless way to setup a mesh with ESP8266 and ESP32 devices
  Website: https://gitlab.com/painlessMesh/painlessMesh
  Category: Communication
  Architecture: esp8266, esp32
  Types: Contributed
  Versions: [0.1.0, 0.1.1, 0.1.2, 0.1.3, 0.1.4, 0.2.0, 0.2.1, 0.5.0, 0.5.4, 0.6.0, 0.6.1, 0.6.2, 0.6.3, 0.6.5, 0.7.0, 1.0.0, 1.0.2, 1.0.3, 1.1.0, 1.1.1, 1.2.0, 1.2.1, 1.2.2, 1.2.3, 1.2.4, 1.2.5, 1.2.6, 1.2.7, 1.2.8, 1.3.0, 1.3.1, 1.3.2, 1.3.3, 1.4.0, 1.4.1, 1.4.2, 1.4.3, 1.4.4, 1.4.5, 1.4.6, 1.4.7]
  Provides includes: painlessMesh.h
  Dependencies: ArduinoJson, TaskScheduler

```
But if i forget the whitespace and search for the header file instead
```zsh
> arduino-cli lib search painlessmesh 
Updating index: library_index.json downloaded                                   
No libraries matching your search.
```
The header file of that library says `painlessMesh.h` but still the search has no match

PR #598 (2020-04-02): Show similar library names in lib search
Closes #107 

Small typos when searching for libraries will show similar library names instead of showing no results.

Emphasis on "small", though, because the similarity value check is hard-coded at a value that's a little strict (maybe this could be a config option?).