arduino/arduino-cli #128: Installing unregistered library

scls19fr, 2019-01-15:
Hello,

It will be nice if arduino-cli could provide an interface to install unregistered library:

From a zip available on HTTP server:

    arduino-cli lib install --unregistered https://github.com/itead/ITEADLIB_Arduino_Nextion/archive/v0.7.0.zip

supporting installing from a git repository might also be considered
(by default it could use master branch but we could install from any given branch, or from a given commit hash, or tag)

You might have a look at how Ruby bundler is doing this
https://bundler.io/v1.12/git.html

Kind regards

BleuLlama, 2019-12-21:
Related; it would be nice to be able to at least use --additional-urls to specify the json files for third party (unregistered) libraries.  (I'd make this another issue/ticket but it seems too related to this one)

ubidefeo, 2020-01-28:
+1 on this
providing a GitHub URL would be ideal, maybe we don't even need to specify flags

asucada, 2020-03-11:
I'll work on this.

alranel, 2020-12-31:
We now have `arduino-cli lib install --git-url <url>` and `arduino-cli lib install --zip-file <url>`. 🎉

closed 2020-12-31 by alranel