arduino/arduino-cli #133: getting default arduino data dir: retrieving user home dir: user: Current not implemented on linux/amd64

zoobab, 2019-01-27:
Hi,

I have made the following Dockerfile to build a blink example for the esp8266 (still WIP):

```
FROM zoobab/arduino-cli
WORKDIR /root
RUN arduino-cli core update-index --debug
RUN arduino-cli sketch new MyFirstSketch --debug
```

If I do a ```docker build```, I end up with the following error:

```
$ docker build .
Sending build context to Docker daemon  43.52kB
Step 1/4 : FROM zoobab/arduino-cli
 ---> 382d5dc8bb65
Step 2/4 : WORKDIR /root
 ---> Using cache
 ---> 8f17febbccda
Step 3/4 : RUN arduino-cli core update-index --debug
 ---> Running in e160743f593c
time="2019-01-27T19:38:18Z" level=error msg="Error creating default configuration" error="getting default arduino data dir: retrieving user home dir: user: Current not implemented on linux/amd64"
Error: getting default arduino data dir: retrieving user home dir: user: Current not implemented on linux/amd64
Error creating default configuration
time="2019-01-27T19:38:18Z" level=error msg="Error creating default configuration" error="getting default arduino data dir: retrieving user home dir: user: Current not implemented on linux/amd64"
The command '/bin/sh -c arduino-cli core update-index --debug' returned a non-zero code: 1
```

After some research I found this discussion:

https://github.com/ksonnet/ksonnet/issues/298

Now if I force the USER as an environment variable, it works fine:

```
FROM zoobab/arduino-cli
WORKDIR /root
ENV USER root
RUN arduino-cli core update-index --debug
RUN arduino-cli sketch new MyFirstSketch --debug
```

Which gives:

```
[...]

Step 4/5 : RUN arduino-cli core update-index --debug
 ---> Running in 3cb3d08ea45b
time="2019-01-27T19:39:49Z" level=info msg="Initiating configuration"
time="2019-01-27T19:39:49Z" level=info msg="Unserializing configurations from /go/bin/.cli-config.yml"
time="2019-01-27T19:39:49Z" level=warning msg="Error reading config, using default configuration" error="open /go/bin/.cli-config.yml: no such file or directory"
time="2019-01-27T19:39:49Z" level=warning msg="Did not manage to get config file, using default configuration" error="open /go/bin/.cli-config.yml: no such file or directory"
time="2019-01-27T19:39:49Z" level=info msg="Checking if CLI is Bundled into the IDE"
time="2019-01-27T19:39:49Z" level=info msg="Candidate IDE Directory: /go/bin"
time="2019-01-27T19:39:49Z" level=info msg="CLI is not bundled into the IDE"
time="2019-01-27T19:39:49Z" level=info msg="Configuration set"
time="2019-01-27T19:39:49Z" level=info msg=arduino-cli-0.3.4-alpha.preview
time="2019-01-27T19:39:49Z" level=info msg="Starting root command preparation (`arduino`)"
time="2019-01-27T19:39:49Z" level=info msg="Formatter set"
time="2019-01-27T19:39:49Z" level=info msg="Updating index" url="https://downloads.arduino.cc/packages/package_index.json"
Updating index: package_index.json downloaded

[...]

```

It seems the environment variable USER has an impact here...

cmaglie, 2019-01-29:
Looks like `user.Current()` do not work if crosscompiled for linux/amd64 or if compiled without CGO enabled.

A possible fix in #137

jpconstantineau, 2019-02-07:
It worked fine on the previous alpha.  I too build a docker image and ran with the same issue with this build.  setting the user ENV variable did work properly as a work around.

SConaway, 2019-03-21:
Same. As a side note, this seems to have been caused between 0.3.3 and 0.3.5

masci, 2019-06-11:
This should be fixed in `master` through #239 can you confirm @zoobab ?

closed 2019-07-17 by masci