arduino/arduino-cli #139: Cannot upload to Arduino Diecimila - not in sync

moustacheful, 2019-01-31:
Followed the instructions on the readme:

```bash
$ arduino core search diecimila 
Searching for platforms matching 'diecimila'

ID         	Version	Name              
arduino:avr	1.6.23 	Arduino AVR Boards

$ arduino core install arduino:avr 
Downloading arduino:avr-gcc@5.4.0-atmel3.6.1-arduino2...
arduino:avr-gcc@5.4.0-atmel3.6.1-arduino2 downloaded                                  
Downloading arduino:avrdude@6.3.0-arduino14...
arduino:avrdude@6.3.0-arduino14 downloaded                                            
Downloading arduino:arduinoOTA@1.2.1...
arduino:arduinoOTA@1.2.1 downloaded                                                   
Downloading arduino:avr@1.6.23...
arduino:avr@1.6.23 downloaded                                                         
Installing arduino:avr-gcc@5.4.0-atmel3.6.1-arduino2...
arduino:avr-gcc@5.4.0-atmel3.6.1-arduino2 installed
Installing arduino:avrdude@6.3.0-arduino14...
arduino:avrdude@6.3.0-arduino14 installed
Installing arduino:arduinoOTA@1.2.1...
arduino:arduinoOTA@1.2.1 installed
Installing arduino:avr@1.6.23...
arduino:avr@1.6.23 installed

$ arduino core list 
ID                	Installed	Latest	Name              
arduino:avr@1.6.23	1.6.23   	1.6.23	Arduino AVR Boards

$ arduino board list
FQBN    Port        	ID       	Board Name
    	/dev/ttyUSB0	0403:6001	unknown

$ arduino compile --fqbn arduino:avr:diecimila Arduino/AnalogReader 
Build options changed, rebuilding all
Sketch uses 930 bytes (3%) of program storage space. Maximum is 30720 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 2039 bytes for local variables. Maximum is 2048 bytes.

$ arduino upload -p /dev/ttyUSB0 --fqbn arduino:avr:diecimila Arduino/AnalogReader
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x00
Error: exit status 1
Error during upload.
```

During the sync checks, the `rx` leds flashes briefly.
This is a very old Arduino Diecimila. Anything I might be missing?

moustacheful, 2019-01-31:
Managed to upload this through the desktop application, but I haven't been able to do this through the CLI, what I was missing was switching the processor to atmega168, but passing it as is yields:

```bash
$ arduino upload -p /dev/ttyUSB0 --fqbn arduino:avr:diecimila:cpu=atmega168 Arduino/AnalogReader
Error: getting build properties for board arduino:avr:diecimila: invalid value 'cpu=atmega168' for option 'cpu'
Invalid FQBN.
```

I'm not sure where to get the exact string for these. Tried atmega168 and ATmega168, to no avail.

per1234, 2019-01-31:
>I'm not sure where to get the exact string for these.

You can get it from boards.txt:
https://github.com/arduino/ArduinoCore-avr/blob/1.6.21/boards.txt#L111-L123
or from enabling **File > Preferences > Show verbose output during > compilation** in the Arduino IDE, compiling something for the desired board configuration, then extracting the FQBN from the first line in the output on the black console window at the bottom of the Arduino IDE window.

That part is related to https://github.com/arduino/arduino-cli/issues/138

>`arduino upload -p /dev/ttyUSB0 --fqbn arduino:avr:diecimila:cpu=atmega168 Arduino/AnalogReader`

Did you rename `arduino-cli` to `arduino`?

`arduino:avr:diecimila:cpu=atmega168` is correct. It works fine for me. I can't imagine what the cause of that `invalid value 'cpu=atmega168' for option 'cpu'` error could be.

scottchiefbaker, 2019-02-01:
This looks like the syntax for the [Arduino IDE CLI](https://github.com/arduino/Arduino/blob/master/build/shared/manpage.adoc) mode. 

Just to add on to what @per1234 said, `arduino:avr:diecimila:cpu=atmega168` is the correct string. The fourth segment is board specific options (like CPU type, baud rate, etc). You can see some other examples in my [Arduino Makefile](https://github.com/scottchiefbaker/Arduino-Makefile) project.

per1234, 2019-05-24:
@moustacheful did you ever find the cause of that error message?:
```
Error: getting build properties for board arduino:avr:diecimila: invalid value 'cpu=atmega168' for option 'cpu'
Invalid FQBN.
```
Have you checked to see if it still occurs with the latest version of arduino-cli?

moustacheful, 2019-05-24:
Hey! sorry, I did not. In the end I just used the gui version, which worked
with no issues.

Sadly, I don't have that arduino with me nor will I have one in the
foreseeable future, so feel free to close this :(

per1234, 2019-05-24:
Thanks for the update @moustacheful. I still can't reproduce the error so I'm going to go ahead and close this issue. If anyone encounters this type of error, please comment here and I can re-open.

closed 2019-05-24 by per1234