arduino/arduino-cli #143: Do not move source files during build

jeff-winn, 2019-02-08:
I'm not entirely sure why the source files are being moved into a temporary folder rather than just referencing their original locations but this does have some impacts to how modern IDEs treat the errors that come out during the compilation process. The code for the screenshots below can be found on my GitHub profile [here](https://github.com/jeff-winn/Nemesis-Mod) if you need a sample.

It could be something you had originally used to service your IDE, which may still be needed. If so, this mechanism should be kept inside *your* IDE and **not** within the CLI. But rather have the CLI simply pointed at your temporary build folder generated by your IDE.

When issues occur within the compilation process many IDEs use the information in the out stream and error stream and parse it out so they can report those issues back to the developer through some easier means of locating them. In this sample I've intentionally misspelt the name of an enum to exhibit what would show up within VS Code.

[image]

You can see from the above screenshot that the compiler has detected the misspelling and reported it back to me. Normally I would simply be able to click on the link and it would take me directly to the file that caused the issue along with move me directly to the offending line to fix it.

[image]

The above error depicts what happens indicating the file was not found (which given it's temporary that's not surprising). However worse would be editing that other copy of the code rather than the real copy that is being managed within source control.

Fixing this and not transferring the code to a separate folder to build there should let the IDEs report back and allow their interfaces to work in the way they were intended.

matthijskooijman, 2020-01-16:
This would be great, but is not quite trivial. The reason files are copied to a build directory, is that all .ino files are concatenated (together with generated forward declaration) and written to a single .cpp file. This file *must* live next to the other sketch files, so it must either be written into the sketch directory (which is impossible, the sketch directory might be read-only and you do not really want to write temp files into sketch directory anyway), or it must be written to a temp build directory and all other files must be copied alongside it.

This requirement is to allow the .ino files to include any source or header files living in the sketch directory. An obvious solution would be to simply add an `-I` option to the sketch directory, but then the difference between `#include "foo.h"` and `#include <foo.h>` breaks. There are some other options that seem useful here, but I was not able to figure out a way to really make this work (I can't remember the details, but IIRC there was a way where includes from the .ino file worked, but then includes from other files broke, or something like that).

Note that the Java IDE might do *another* copy of your sketch if you have saved changes, to allow compiling your unsaved changes without overwriting any data.

A somewhat (but not quite) related issue is https://github.com/arduino/arduino-builder/issues/86

jeff-winn, 2020-01-20:
Unfortunate but not unexpected, going to close.

closed 2020-01-20 by jeff-winn