arduino/arduino-cli #146: minicore and compile options

a7hybnj2, 2019-02-12:
I managed to install [minicore](https://github.com/MCUdude/MiniCore) but I don't see the options needed to compile correctly.
```
Board:
Bootloader:
Clock:
BOD:
Compiler:
Variant:
Port
```
a screenshot from the IDE
[image]

per1234, 2019-02-12:
It sounds like a duplicate of https://github.com/arduino/arduino-cli/issues/138

In that issue, I have provided instructions for how you can determine the FQBN for any board. This is the solution I would recommend for you:

>My current approach in the case of boards with custom Tools menus is to do a compilation in the Arduino IDE for the specific board configuration I want with **File > Preferences > Show verbose output during: compilation** enabled, and then copy the FQBN from the first line of the output in the black console window at the bottom of the Arduino IDE window. Quite crude, but it seems the fastest and least error-prone solution for me.

If I misunderstood and you are reporting a bug in arduino-cli or submitting a feature request here, then please provide more details and I'll be happy to re-open the issue. If you need assistance using MiniCore with arduino-cli, please post on the [Arduino Forum](http://forum.arduino.cc/) and we'll help you out over there.

closed 2019-02-12 by per1234

a7hybnj2, 2019-02-12:
@per1234 Thanks!
` -fqbn=MiniCore:avr:328:bootloader=false,variant=modelP,BOD=2v7,LTO=Os,clock=16MHz_external `

pouriap, 2022-02-10:
>  I guess I will have to type that in every time!? yikes.

Since this is the first result in Google for "arduino-cli minicore", in case you want certain options to stick to a sketch you can use the "attach" command:

```
arduino board attach -b MiniCore:avr:8:bootloader=no_bootloader,BOD=4v0,LTO=Os_flto,clock=8MHz_internal "PATH\TO\PROJECT" --port COM3
```

Now every time you compile or upload these settings will be used by default, unless you override them.