arduino/arduino-cli #150: config file doesn't work

berry0511, 2019-02-18:
version:0.3.4
platform:win10 x64

when i set root path in the file arduino-cli.yaml, it doesn't work:
Windows:
   - paths:
     - sketchbook_path: ./
     - arduino_data: ./
   - proxy:
     - type: auto

the cli tool download path is still in C:\Users\***\AppData\Local

cmaglie, 2019-02-20:
yep I'm about to release 0.3.5 that fixes this.

cmaglie, 2019-02-20:
Please re-download the file now, it should be available the version 0.3.5-alpha.preview. Reopen the issue if the problem persists.

SebCanet, 2019-02-22:
With 0.3.5 version I have the same pb.
Inside mine:
```
sketchbook_path: ../sketchbook
arduino_data: ./
board_manager:
  additional_urls:
    - http://arduino.esp8266.com/stable/package_esp8266com_index.json
    - https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json
    - https://dl.espressif.com/dl/package_esp32_index.json
    - http://digistump.com/package_digistump_index.json
```

cmaglie, 2019-02-22:
Please tell me the full path of:
- `arduino-cli` executable
- `arduino-cli.yaml`

and copy and paste the output with the `--debug` flag.

SebCanet, 2019-02-22:
executable: F:\Logiciels\Arduino_graphique\BlocklyArduino_electrified\arduino
yaml: F:\Logiciels\Arduino_graphique\BlocklyArduino_electrified\arduino

# arduino-cli core update-index --debug
INFO[0000] Checking if CLI is Bundled into the IDE
INFO[0000] Candidate IDE Directory: F:\Logiciels\Arduino_graphique\BlocklyArduino_electrified\arduino
INFO[0000] CLI is not bundled into the IDE
INFO[0000] Configuration set
INFO[0000] arduino-cli-0.3.5-alpha.preview
INFO[0000] Starting root command preparation (`arduino`)
INFO[0000] Formatter set
INFO[0000] Updating index                                url="https://downloads.arduino.cc/packages/package_index.json"
Updating index: package_index.json downloaded

PR #154 (2019-02-22): bugfix: Navigate method didn't work with Windows-style paths
Fix #150 

/cc @berry0511 @SebCanet
commit 9e0a1f84 (2019-02-22): bugfix: Navigate method didn't work with Windows-style paths

Fix #150
closed 2019-02-22 by cmaglie