arduino/arduino-cli #153: cannot install arduino:stm32f4

matteosuppo, 2019-02-21:
```
$ ino core install arduino:stm32f4
Tool arduino:arm-none-eabi-gcc@4.8.3-2014q1 already installed
Tool arduino:dfu-util@0.9.0-arduino1 already installed
Tool arduino:arduinoSTM32load@2.0.0 already installed
arduino:stm32f4@1.0.1 already downloaded
Installing arduino:stm32f4@1.0.1...
Error: searching package root dir: no unique root dir in archive, found '/home/builder/.arduino15/tmp/package-687061712/cores' and '/home/builder/.arduino15/tmp/package-687061712/drivers'
Cannot install platform
```

I don't know if it's cli bug or a core bug

e76nystrom, 2019-03-30:
I get a similar error running on Windows 10 64 bit

arduino-cli core install arduino:stm32f4
Tool arduino:dfu-util@0.9.0-arduino1 already installed
arduino:arm-none-eabi-gcc@4.8.3-2014q1 already downloaded
arduino:arduinoSTM32load@2.0.0 already downloaded
arduino:stm32f4@1.0.1 already downloaded
Installing arduino:arm-none-eabi-gcc@4.8.3-2014q1...
arduino:arm-none-eabi-gcc@4.8.3-2014q1 installed
Installing arduino:arduinoSTM32load@2.0.0...
arduino:arduinoSTM32load@2.0.0 installed
Installing arduino:stm32f4@1.0.1...
Error: searching package root dir: no unique root dir in archive, found 'C:\Users\Eric\AppData\Local\Arduino15\tmp\package-189434413\cores' and 'C:\Users\Eric\AppData\Local\Arduino15\tmp\package-189434413\drivers'
Cannot install platform

It works okay from the gui after the cli has failed.  The cli then shows:

arduino-cli core install arduino:stm32f4
Platform arduino:stm32f4@1.0.1 already installed

masci, 2019-08-06:
Moving discussion to https://github.com/arduino/arduino-cli/issues/325

closed 2019-08-06 by masci