arduino/arduino-cli #155: bootloader gets destroyed every time

1inguini, 2019-02-25:


per1234, 2019-02-26:
Which version of arduino-cli are you using?

Please provide the arduino-cli commands you are using to compile and upload.

1inguini, 2019-02-26:
command: probably `➤ arduino-cli upload -p /dev/ttyACM0 --fqbn SparkFun:avr:promicro Arduino/FirstSketch/` accrding to my `fish`'s history.

version: `arduino-cli version 0.3.4-alpha.preview`

closed 2019-11-23 by github-actions[bot]

bjanders, 2021-03-17:
I have the same issue, also with a SparkFun pro micro clone. I used a self compiled arduino-cli, commit 550179ee

ubidefeo, 2021-03-18:
@bjanders when you compile for a target board you also build a version of the firmware with included bootloader.
This is the one you should burn (upload) onto the board.

essentially it becomes a two part operation in which you compile choosing to `--export-binaries`, 
thus creating a local `build` folder inside your sketch, and then `upload` with `--input-file` and passing the `build/SparkFun.avr.micro/MySketch.ino.with_bootloader.hex`

unfortunately I seem to have found an issue with this particular file in my case, which might be related to an old problem we had with CLI assembling the bootloader binary with the newly created hex.

@cmaglie and I are going to investigate asap

here's a recap of my actions for a sketch named `BlinkMe`.
I'm inside the `BlinkMe` folder

```
arduino-cli compile --export-binaries -b arduino:avr:micro -v
arduino-cli upload -b arduino:avr:micro -v -P arduinoisp --input-file build/arduino.avr.micro/BlinkMe.ino.with_bootloader.hex
```

bjanders, 2021-03-21:
Thanks for the information! Just to make sure I understood you correctly: Does the method using `--export-binaries` and `--input-file` work around the issue you have found, so this will work and that is what I should use for now?

What I did was according to https://arduino.github.io/arduino-cli/latest/getting-started/:
```
arduino-cli compile -b SparkFun:avr:promicro
arduino-cli upload -p COMx -b SparkFun:avr:promicro 
```
Will this work when the issue has been fixed?

ubidefeo, 2021-03-21:
@bjanders 
this discussion is related to using a programmer to upload the firmware, so you'll have to pass in `-P` and the specific programmer supported by the platform.
It should work after the files are correctly combined, but not sure when this will happen right now.

I remember we fixed it for the SAMD21 a while back, but I need someone to look into this

bjanders, 2021-03-22:
I'm sorry, I'm not able to follow what you mean. Which discussion is related to using a programmer are you referring to?

The problem I have, i.e the bootloader getting corrupted manifests itself when I'm not using a programmer, but flashing using the USB bootloader. And it does not manifest itself when I flash with_bootloader.hex when I'm not using a programmer, but again using the USB bootloader. So in my case the issue is never related to a programmer.

bjanders, 2021-03-22:
It seems I was wrong. Flashing with the above working command only worked as long as kept it powered.