arduino/arduino-cli #162: too many files, too many folders

SebCanet, 2019-03-05:
Hi,
I use your great tool for my project, but on Windows we have a problem with too long filname if the user copy your soft in some folder with long name or subfolders.

Could it be possible, when you download a core, to add a 'compress' option? Like you have on Electron with this (strange) asar file, which inside are all node_modules files.

thanks.

mastrolinux, 2019-09-18:
Please look into: https://docs.microsoft.com/en-us/windows/win32/fileio/naming-a-file#enable-long-paths-in-windows-10-version-1607-and-later

And with Git:
`git config --system core.longpaths true`

closed 2019-09-18 by mastrolinux

SebCanet, 2019-09-18:
You cannot ask to a user to modify (and know how) his OS system !?
What about Win XP/7/8 users?
And even with other filesystem, the gcc compiler has too many files in too many folders, I cannot copy itquickly on USB key for my students.
Ths problem is still opened.

facchinm, 2019-09-19:
@SebCanet paths handling on windows is always an issue; we tried to overcome some of these problems by reducing the archived core filename (https://github.com/arduino/arduino-builder/commit/ab41198a9a3a9da3e04ca493fd245d0e87b4785c) and the commandline length (https://github.com/arduino/arduino-cli/blob/84172d8902fcd89d71f1dc6507c6af3daa3f274c/legacy/builder/builder_utils/utils.go#L525) but there's still a corner case here and there.