arduino/arduino-cli #163: Compilation fails for ESP32 : myproject.partition.bin missing.

UltimateUnicorn, 2019-03-06:
Hello. Thanks for the support and many advices you've already provided here.
I would like to report a bug that occurs when I try to compile a sketch for an ESP32.

### My setup
- Ubuntu 18.04.2 LTS 64 bits
- Arduino-cli version 0.3.6-alpha.preview
- ESP32 boards core version 1.0.1 (added [https://dl.espressif.com/dl/package_esp32_index.json](https://dl.espressif.com/dl/package_esp32_index.json))
- Espressif ESP32 DevKitC (module : ESP32-WROOM-32D)

### Steps to reproduce
- Create a new project (I copied the _HelloServer.ino_ from the examples directory into a new directory called _test_ and renamed _HelloServer.ino_ into _test.ino_).
- Compile it using `arduino-cli compile --fqbn esp32:esp32:esp32 test` (from the parent directory of _test_) -> works well (I got a "normal" output from the compiler and the following files have been created in the _test_ folder : _test.esp32.esp32.esp32.bin_ & _test.esp32.esp32.esp32.elf_.
- Upload the project : `arduino-cli upload --fqbn esp32:esp32:esp32 -p /dev/ttyUSB0 test` -> doesn't work. I get the following output :

> No new serial port detected.
> usage: esptool write_flash [-h] [--flash_freq {keep,40m,26m,20m,80m}]
>                            [--flash_mode {keep,qio,qout,dio,dout}]
>                            [--flash_size FLASH_SIZE]
>                            [--spi-connection SPI_CONNECTION] [--no-progress]
>                            [--verify] [--compress | --no-compress]
>                            <address> <filename> [<address> <filename> ...]
> esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: 'test/test.esp32.esp32.esp32.partitions.bin'
> Error: exit status 2
> Error during upload.

### Additional information
- When using the Espressif toolchain & ESP-IDF on the same computer I can upload the sketch and it runs.
- I used the exact same procedure using the Arduino IDE version 1.8.8 under Windows : it worked flawlessly.
- I just realized that there is no newline between directory: and 'test/tes....." on my console.

Feel free to contact me for additional details / testings.
Thanks.

per1234, 2019-03-07:
The ESP32 core's `recipe.objcopy.eep.pattern` creates a file named `{build.project_name}.partitions.bin`:
https://github.com/espressif/arduino-esp32/blob/master/platform.txt#L80
```
recipe.objcopy.eep.pattern={tools.gen_esp32part.cmd} -q "{runtime.platform.path}/tools/partitions/{build.partitions}.csv" "{build.path}/{build.project_name}.partitions.bin"
```
and then the upload recipe expects to find that file in `{build.path}`, but arduino-cli doesn't copy that file from the temporary build folder to the sketch folder, thus the error.

WayneBooth, 2019-03-30:
I have same problem. Happy to provide details is required.
For now I have two **workarounds**:

Edit platform.txt to remove the upload for partitions.bin
102,103c102,105
< tools.esptool_py.upload.pattern="{path}/{cmd}" --chip esp32 --port "{serial.port}" --baud {upload.speed}  --before default_reset --after hard_reset write_flash -z --flash_mode {build.flash_mode} --flash_freq {build.flash_freq} --flash_size detect 0xe000 "{runtime.platform.path}/tools/partitions/boot_app0.bin" 0x1000 "{runtime.platform.path}/tools/sdk/bin/bootloader_{build.boot}_{build.flash_freq}.bin" 0x10000 "{build.path}/{build.project_name}.bin" 0x8000 "{build.path}/{build.project_name}.partitions.bin"
< tools.esptool_py.upload.pattern.linux=python "{path}/{cmd}" --chip esp32 --port "{serial.port}" --baud {upload.speed}  --before default_reset --after hard_reset write_flash -z --flash_mode {build.flash_mode} --flash_freq {build.flash_freq} --flash_size detect 0xe000 "{runtime.platform.path}/tools/partitions/boot_app0.bin" 0x1000 "{runtime.platform.path}/tools/sdk/bin/bootloader_{build.boot}_{build.flash_freq}.bin" 0x10000 "{build.path}/{build.project_name}.bin" 0x8000 "{build.path}/{build.project_name}.partitions.bin"
---
> tools.esptool_py.upload.pattern="{path}/{cmd}" --chip esp32 --port "{serial.port}" --baud {upload.speed}  --before default_reset --after hard_reset write_flash -z --flash_mode {build.flash_mode} --flash_freq {build.flash_freq} --flash_size detect 0xe000 "{runtime.platform.path}/tools/partitions/boot_app0.bin" 0x1000 "{runtime.platform.path}/tools/sdk/bin/bootloader_{build.boot}_{build.flash_freq}.bin" 0x10000 "{build.path}/{build.project_name}.bin" 
> #0x8000 "{build.path}/{build.project_name}.partitions.bin"
> tools.esptool_py.upload.pattern.linux=python "{path}/{cmd}" --chip esp32 --port "{serial.port}" --baud {upload.speed}  --before default_reset --after hard_reset write_flash -z --flash_mode {build.flash_mode} --flash_freq {build.flash_freq} --flash_size detect 0xe000 "{runtime.platform.path}/tools/partitions/boot_app0.bin" 0x1000 "{runtime.platform.path}/tools/sdk/bin/bootloader_{build.boot}_{build.flash_freq}.bin" 0x10000 "{build.path}/{build.project_name}.bin" 
> #0x8000 "{build.path}/{build.project_name}.partitions.bin"

or simply upload manually
path_to_tool/esptool.py --port /dev/ttyUSB0 write_flash 0x10000 ~/Arduino/tst/tst.esp32.esp32.esp32.bin

ubidefeo, 2019-04-27:
@WayneBooth I recommend manual upload for now.
If you issue an `erase_flash` command this will delete the partition scheme and if you manually upload a firmware after such operation your program will not run.
@cmaglie how can we tackle copying the partition.bin file as well?

thbar, 2019-07-26:
I met the same problem. Thanks @WayneBooth for the work-arounds. If it helps someone not familiar with esptool, on a Mac here is exactly what I had to do (using the installed Arduino IDE):

```
~/Library/Arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool --port /dev/cu.SLAB_USBtoUART write_flash 0x10000 WifiScan/WifiScan.esp32.esp32.esp32.bin 
```

After that, attaching using `screen /dev/cu.SLAB_USBtoUART 115200` allowed me to verify that the upload succeeded.

jdwolin, 2019-08-05:
Same problem here!

commit 09e3c42e (2019-10-29): Copy multiple build artifacts matchin a name pattern.

Previously the compile command copied only the following artifacts:

  sketch.ino.hex (or .bin)
  sketch.ino.elf

now also the files matching the glob sketch.ino.*.hex will be copied.
This allows for some 3rd party core (like esp32) to copy also the extra
file sketch.ino.partitions.bin.

Should fix #163
PR #457 (2019-10-29): Fixed upload issue on esp32 (partitions artifacts copy)
Previously the compile command copied only the following artifacts:

```
sketch.ino.hex (or .bin)
sketch.ino.elf
```

now also the files matching the glob

```
sketch.ino.*.hex
```

will be copied.
This allows for some 3rd party core, like esp32, to copy also extra files like `sketch.ino.partitions.bin`.

Should fix #163
commit e51df0d1 (2019-10-29): Fixed upload issue on esp32 (partitions artifacts copy) (#457)

* Copy multiple build artifacts matching a name pattern.

Previously the compile command copied only the following artifacts:

  sketch.ino.hex (or .bin)
  sketch.ino.elf

now also the files matching the glob sketch.ino.*.hex will be copied.
This allows for some 3rd party core (like esp32) to copy also the extra
file sketch.ino.partitions.bin.

Should fix #163

* Added test for improved support for esp32:esp32

https://github.com/arduino/arduino-cli/issues/163

* Run of "go mod tidy"
closed 2019-10-29 by cmaglie